omersiar/esp-rfid #99: esp-rfid Does not boot anymore when upgraded to 0.7

omersiar, 2018-04-02:
There is a known issue with the configuration file when you upgrade the firmware from any version to 0.7.x+

To workaround this issue

* Make sure you have a user's backup ( do not back up settings )
* Before upgrading make a factory reset.
* After factory reset connect to esp-rfid's Access Point.
* Do not setup any configuration
* Do firmware upgrade
* Configure your newly upgraded device
* Save and Reboot
* Restore Users back.

pidiet, 2018-04-08:
True.

Just tried an update from web ui from 0.7.0 to 0.7.2 (without any prerequisites) and my device was blowing away...

omersiar, 2018-04-13:
Not modified header will be dropped, since the firmware actively changing, we do not need it for now.  It can be enabled again when we hit the v1.

closed 2018-04-13 by omersiar