omersiar/esp-rfid #101: permanent access point mode

teriall, 2018-04-04:
After I rebuild the project by myself in platformio the access point mode become permanent, even it's client mode in setting and actually it's working both modes (is it possible?). In terminal I getting log that the module was connected to the network, but the access point still working as well. I didn't change much just reassemble for dio mode.

omersiar, 2018-04-04:
It is possible but in rare cases, for example previous firmware can cause that, i suggest using erase.bat to format whole flash so your esp will act as desired.

Bluebie, 2018-04-11:
I'm also having this issue. Whenever it's placed in to Client (station) mode in the webui settings, it goes in to dual mode and broadcasts an access point as well as connecting as a station. If i switch to Access Point mode i can change the network name and add a password, but it will still create the network with those new settings when i set it back to Client mode in the webui and reboot. This seems like a bug. A reasonable firmware should configure the wifi mode properly.

omersiar, 2018-04-11:
Thank you for the feedback, now i consider this as a bug. Probably this is the main cause https://github.com/omersiar/esp-rfid/blob/stable/src/main.cpp#L996

teriall, 2018-05-09:
Any idea how to solve it? :)

omersiar, 2018-05-09:
@teriall Did you try latest version?

teriall, 2018-05-09:
just downloaded it - still same, version 0.7.4, on status page version 0.7.3

allserwis, 2018-05-31:
I have the same issue. Tried 0.74 and 0.73 on two NodeMCU modules and still the same. Permanent AP. I tried everything, blank flash, different configurations etc. Still without success :-( 
With ver 0.74 two times after long time get access from LAN, but after reboot back to AP and keep rebooting :-(
Plus WiFi reception is much worse than on ESPeasy. Both modules work strait with ESPeasy.

closed 2018-06-09 by omersiar

alegiovannelli, 2018-06-09:
Hi, this issue has been closed, but I have the same problem too. Was the solution found? Thanks, great project.

alegiovannelli, 2018-06-10:
Yes, yesterday i tried new version and now seems to be ok! Thank you very much.

allserwis, 2018-06-27:
Unfortunately is a still the same :-(

teriall, 2018-06-28:
with the latest release don't have this bug anymore

rpanfili, 2018-09-09:
I got the same bug on v0.8.2. I configured my wireless network without any BSSID specified (I got multiple AP) and I was looping in this kind of issue without being able to let the esp join my network. I specified the BSSID and everything goes fine. I then removed the BSSID, saved and rebooted and now I got no issues anymore!

spatnynick, 2019-09-19:
I had with the problem too, v. 1.3.1. I prepared my rf-id in basement, everything worked well, I installed it in garden and no wifi client, only AP mode. After some investigation I found out that the problem probably is related to in infrastructure, I have more APs with the same SSID, but different BSSID. Then also WifiScan result is strange - I had some of them without SSID in the resultlist.
If I set only SSID, no BSSID I got only IP mode... one would expect that it chooses the AP is currently the strongest signal