omersiar/esp-rfid #102: Receive MQTT Commands

Schnittcher, 2018-04-06:
It would be nice if you could execute commands via MQT to protect the relay. Is that planned?

omersiar, 2018-04-08:
Can you share how can it be used?

Schnittcher, 2018-04-08:
Yes, i want to integrate it in our Smart Home, and I want to implement the user accesses there.

quikote, 2018-04-08:
+1
I would like to operate de door from Home Assistant using MQTT, so I need to send a MQTT message from my home controller to activate the lock as a allowed RFID card was readed

Schnittcher, 2018-04-08:
I found the testrelay command, so I can control the relay. Can you still tell me how I can read the user list and add user?

omersiar, 2018-04-10:
`"command": "userfile"` is the for adding new user.

omersiar, 2018-04-13:
Did you managed to send WebSocket commands via Home Assistant?

quikote, 2018-04-13:
I don't how to.... :-(((

omersiar, 2018-04-13:
Currently we have WebUI to setup everything. MQTT was never an intended feature to support, It will take longer to support incoming mqtt commands. Closing this for now.

closed 2018-04-13 by omersiar