omersiar/esp-rfid #103: Publish mqtt availability topic of esp-rfid

rradar, 2018-04-09:
Would be really nice to see esp-rfid publishing it's availability topic.

For home assistant this information is really nice! Could be something like this (ha yaml code entry):

```
availability_topic: 'esp-rfid/status'
payload_available: 'online'
payload_not_available: 'offline'
```

When the device is offline (network wise) I could see this information in home assistant :+1:

magic-tune, 2018-04-11:
How should that work? When the device is offline it can`t publish anything.
You could use home assistant ping sensor for that.

quikote, 2018-04-11:
+1
I would also like to see this feature called LWT (Last Will and Testament) implemented in esp-rfid. In this way we will see in the controller (Home Assistant for example) if the device is online or not. I usually use it with every MQTT client connected to HA. 

The Last Will and Testament (LWT) feature is used in MQTT to notify other clients about an ungracefully disconnected client. Each client can specify its last will message (a normal MQTT message with topic, retained flag, QoS and payload) when connecting to a broker. The broker will store the message until it detects that the client has disconnected ungracefully. If the client disconnect abruptly, the broker sends the message to all subscribed clients on the topic, which was specified in the last will message. The stored LWT message will be discarded if a client disconnects gracefully by sending a DISCONNECT message.

rradar, 2018-04-11:
@magic-tune -> A picture says more than thousand words:

[image]

All sensor are **offline**. Just the upper one (Multisensor) supports the 'availability' topic. The lower one just shows his **last** values which could be a year old (because the device could offline since a year...)

magic-tune, 2018-04-12:
Ah ok. I did'n know about this testament thing. So when the esp-rfid connects, it say to the broker: “hey when i die, push this message to my friends.“ When the broker realized that esp-rfid is died, then a message will be pushed to all subscribers “esp-rfid is death! You can't enter your home any more“. What happens when the broker goes down? Why not to use the ping-plugin for that ? 
Asynchron mqtt supports this testament messages.
But I do not see the advantage of this in esp-rfid. Maybe when you hook up some 24/7 sensors?
What does a rfid cardreader do all day long? Right 99,9% nothing. So why it should be connected to wifi and consume power all day long? I hooked my to a pir like [this](https://m.de.aliexpress.com/item/32832812101.html?trace=wwwdetail2mobilesitedetail&productId=32832812101&productSubject=LED-Strip-Light-Motion-Sensor-Switch-12V-24V-5V-Movement-Sensor-Motion-Automatic-Infrared-Timer-Detector&spm=a2g0s.9042311.0.0.JEWkuW). So when there is movement, esp-rfid gets power, connects to wifi/broker (by the way this happens very fast. I test it for a week now with no issues at all) and when the right card is present it activates the relay, push a mqtt message. When there is no movement for 2 minutes, the pir cuts the power. For now this is the best solution 4 me. Dont know if the powercuts effect the esp health? I will see.
Maybe connecting a pir to gpio and using deepsleep is a better option for the future?

cheers 
magic-tune

pidiet, 2018-04-13:
@magic-tune for your environment the LWT gives you even more information than for someone you has esp-rfid running as a 24-7 always-on node. But even with 24-7 the information of a disconnected device is really helpful and makes debugging much easer!

This for example would be a 24-7 always-on node **OR** your esp-rfid which just comes online when needed

[image]
_shows always the last uid (No matter if online or offline)_
 
This one could be your esp-rfid LWT. Actually you can utilized your information (online/offline) from your esp and create a binary-sensor out of it. So even you PIR is not even connected to your home assistant it can receive the information of motion (little bit of topic so)...
Anyway you could see with LWT in the history when esp-rfid was active (online) in your case which is when the pir activated the device. Also the time frame when the tag was used is visible (missing right now because 'always' the last uid is shown.

[image]
_shows the last uid (on the little green field) **and** online (on) or offline (off) state_ 

And pinging is not a good idea btw. Needs more energy (on two devices!) and flooding the network with unnecessary traffic. It's also more dumb (just hitting every n seconds/minutes with the hammer) and not smart like a LWT.

Pablo2048, 2018-04-13:
> And pinging is not a good idea btw. Needs more energy (on two devices!) and flooding the network with unnecessary traffic. It's also more dumb (just hitting every n seconds/minutes with the hammer) and not smart like LWT.

Sorry, but this is definitely not true - how do you think MQTT knows when client die? By using keep-alive packets, so there is always some traffic and the two devices have to be online...

pidiet, 2018-04-13:
> Sorry, but this is definitely not true - how do you think MQTT knows when client die? By using keep-alive packets, so there is always some traffic and the two devices have to be online...

If it's _inside_ the mqtt protocol it should be good without any downsides (it's by design a low traffic protocol/architecture).
I thought more about using nmap and scanning your network at given time intervals (because there exists plugins for home assistant) for certain targets. If you will test it on battery powered nodes/devices you will see a heavy increase of battery usage on these devices. You can test it easily with your phone and look in the battery stats afterwards - deep sleep will permanently be interrupted.

So my last will and testament: **we all need LWT for esp-rfid** :+1:

omersiar, 2018-04-13:
WebSocket messages can be used out of the box.  You will need to wait for MQTT support.

For LWT just tell which message to publish.

Also I think Async MQTT Library is not as stable as PubSubClient.

quikote, 2018-04-17:
Normally my sensors sent a "Online" or "Offline" message as LWT.
I just have to listen to a topic, as for example 'stat/<TOPIC NAME>/LWT' for "Offline" or "Online" message.

rradar, 2018-04-23:
I think this one here could be used as a starter:

```
/*
  Function called to connect/reconnect to the MQTT broker
*/
void connectToMQTT() {
  if (!mqttClient.connected()) {
    if (lastMQTTConnection < millis()) {
      if (mqttClient.connect(MQTT_CLIENT_ID, MQTT_USERNAME, MQTT_PASSWORD, MQTT_AVAILABILITY_TOPIC, 0, 1, "offline")) {
        DEBUG_PRINTLN(F("INFO: The client is successfully connected to the MQTT broker"));
        publishToMQTT(MQTT_AVAILABILITY_TOPIC, "online");
```

Source: https://github.com/mertenats/Open-Home-Automation/blob/master/ha_mqtt_multisensor/ha_mqtt_multisensor.ino

omersiar, 2018-07-18:
https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#asyncmqttclient-setwillconst-char-topic-uint8_t-qos-bool-retain-const-char-payload--nullptr-size_t-length--0

omersiar, 2018-08-03:
There will be LWT on next release, closing it now, track it on #125

closed 2018-08-03 by omersiar