omersiar/esp-rfid #105: Nothing works... PN532

JuculianD, 2018-04-14:
After I tried it with 2 Wiegand Reader, 5 wemos d1 mini, I came to the conclusion that only the MFRC reader worked. I now want to use the PN532 but unfortunately, I got many problems.
The wemos d1 mini does not want to start with the PN532 connected and often, it needs several attempts to flash the Wemos D1 Mini with the included script.

Can someone please help? It is urgent.

JuculianD, 2018-04-17:
Thank you very much! I will use the MFRC522 for now. Do you think that I can increase the reading range a bit, like up to 6-7cm?

omersiar, 2018-04-17:
Did you try max gain setting on mfrc522?

JuculianD, 2018-04-18:
yes

omersiar, 2018-04-28:
@JuculianD You will need better antenna design to for better range.

closed 2018-04-28 by omersiar