omersiar/esp-rfid #107: Access Type Selection

araziforum, 2018-04-19:
[image]
Hi. In first thx for your project. I want to ask you about "Access Type" menu. I choose "Within Selected Hours" but how can I edit or change this hours settings?

omersiar, 2018-04-19:
Hello @araziforum,

I taught that i could quickly implement these settings but i am afraid that the most of the access type option currently not implemented.  Sorry for the inconvenience,  i should remove them until it's done.

araziforum, 2018-04-19:
Thx for your very fast response. I hope you will succeed in your work. Have a nice day.

araziforum, 2018-04-20:
@omersiar Hocam anladığım kadarıyla Türksünüz. Bunun Türkçe olanından yapmak mümkün müdür?

closed 2018-04-23 by araziforum