omersiar/esp-rfid #108: Constantly rebooting

alfredtorch, 2018-04-19:
Hi,

I tried now two different nodemcu (v2-v3). Connected with the default pin to a RC522. Once, I want to save my configuration, (hardware or wifi), it reboots, trying to connect to the registered Wifi AP and keeps crashing with restarting. I tried also to change the board definition (changing board to nodemcuv2 and adapting the memory in the platformio.ini, without any luck). It's a new setup, I did also the blank bin upload. What I'm I missing? The troubles were quite the same than the previous reported crashes on github.

Thank you for your help,

Best regards,

Max

```
[ INFO ] Client IP address: 10.1.1.198

Exception (28):
epc1=0x4021ea20 epc2=0x00000000 epc3=0x00000000 excvaddr=0x05ff907b depc=0x00000000

ctx: cont 
sp: 3ffffc00 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffda0:  3fff05b4 00000007 00000003 40219a52  
3ffffdb0:  3fff0033 00000000 3ffffe10 0000092f  
3ffffdc0:  00000010 3ffffe50 3ffffe50 4021a14f  
3ffffdd0:  3fff29ac 00000006 3ffffe50 4021a19b  
3ffffde0:  3ffe9b2f 3fff05b4 3ffffe50 4021a1cd  
3ffffdf0:  4020c5e7 3ffffe20 3fff04b0 4021a21a  
3ffffe00:  05ff907b 00ffffff 0101010a 402199ac  
3ffffe10:  3fff29ac 3fff04b0 3fff05b4 40234344  
3ffffe20:  4021d088 c601010a 3ffffe98 4021a14f  
3ffffe30:  3fff319c 00000005 3ffffe98 4021a19b  
3ffffe40:  3fff089c 0000042d 0000042d 4010020c  
3ffffe50:  3fff331c 0000000f 00000006 40208ed0  
3ffffe60:  3fff319c 3fff2cc0 3ffffe98 3fff29ac  
3ffffe70:  3fff04b0 3fff2cc0 3ffffe98 402357ca  
3ffffe80:  4021d030 00000000 000003e8 00000008  
3ffffe90:  3fff291c 3fff295c 00000000 00000000  
3ffffea0:  00000000 feefeffe 4021cc68 00000020  
3ffffeb0:  3fff2dac 00000400 3ffffe98 0000031f  
3ffffec0:  3fffff00 3fff05b4 3fff2020 3fff239c  
3ffffed0:  3fff2034 4021b845 00000578 00000578  
3ffffee0:  000000f0 3fff1bf8 3fff1bf4 4021b86d  
3ffffef0:  00000578 4021cbac 3fffff8c 4020c145  
3fffff00:  4021b9e4 4021baa0 88bc2500 000fed1d  
3fffff10:  3fff2974 00001000 00002000 00000100  
3fffff20:  3fff29c7 3fff2e90 3fff2e38 3fff2ac2  
3fffff30:  0000001e 3fff1f2c 00000000 3fff0674  
3fffff40:  00000000 3fff1bf8 3fff1bf4 4021b8b7  
3fffff50:  3fffdad0 00000000 3fff05b4 3fff0674  
3fffff60:  3fff088c 3ffe95e7 3ffe90fc 40235a80  
3fffff70:  40208208 feefeffe 4021be04 4021bdf0  
3fffff80:  00000000 00000000 feefeffe 00000000  
3fffff90:  00000000 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3fff066d 4021acd0  
3fffffc0:  feefeffe feefeffe 3ffe857c 40100721  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

```

omersiar, 2018-04-20:
I need .elf file to debug.

alfredtorch, 2018-04-24:
I got it working, with a Wemos D1 finally. I'll try your approach and report later on it. Thank you.

alfredtorch, 2018-04-28:
Only using a Wemos D1 and the magic happened, flashing with esptool with the same command as below. However, I got 2 new issues, with the wifi setup at the final location of the project. I didn't had this problem at my home using the same router. So now, it's running perfectly, with the default wifi settings. In any case thank you for sharing your programming talents!

1st issue: Once trying to connect as client WPA2 Personnal Network (20chars long password) 
```
[ INFO ] Trying to connect WiFi: CampingsStuff....
[ INFO ] Client IP address: 192.168.178.47

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

2nd issue: After I was trying to setup as the esp as AP (SSID KeyBox - short password)

```
[ INFO ] ESP-RFID is running in AP Mode 
[ INFO ] Configuring access point... Failed!

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

ESPEOL, 2018-04-28:
@omersiar It changes something considering the Watchdog Timer.

@alfredtorch The first one: It also means, the WatchDog Timer had run out, so the esp restarted with booting from spi flash
Just do as I mentioned earlier, 3V to D1.

The second one reffers to the reset pin, so please check, if it is not connected anywhere (your finger might be enough).
What I‘ve experienced myself, it could also mean your used power supply can‘t handle the current demand (which is quite high at the begining, USB2.0 and a bad cable can cause this instability. If you can, buy (or make) your own 5V power supply, mine consists of a 9 V supply, I had laying around and a 5V regulator. This could output 10 A so its more than enough. 

In a nutshell, the first one can be solved by bridging 3 V and D1, the second one by using a better power supply.

And for all who want to know why the esp i drawing so much current at the beginning: 
The card reader and the relay get initialised at the same time, and during that the not small program also gets loaded. This easily exceeds 200 mA, if not 300.
And by using a bad USB cable, and using USB 2.0 (which can only output 300 mA) problems can be plannend. 

I‘d be happy to hear, if I could solve your problem, or if not, how you fixed it.