omersiar/esp-rfid #114: About adding new hardware

araziforum, 2018-04-29:
Hi. Can you add a buzzer controller pin in hardware menu? (Hocam kartın sistemi çalıştırdığını anlamak için buzzer eklenmesi mümkün mü? Bu güzel bir ek özellik olacaktır. Röleyi seçtiğimiz gibi buzzer için de bir pin seçilmesi mümkün olur mu?)

omersiar, 2018-04-30:
Merhaba,

Buzzer veya led eklenmesi için henüz erken.  Kararlı bir sürüm çıkarana kadar eklemeyi planlamıyorum, daha doğrusu öncelikli değil.

omersiar, 2018-04-30:
@chaintip

chaintip, 2018-04-30:
`0.00358035 BCH`| `~   4.97 USD` has [been tipped](https://explorer.bitcoin.com/bch/address/bitcoincash:qqh0ld4v2fsutwt6c4n932w4j6wy20emkucv2nyqw3) to this issue by @omersiar.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/30), get a pull request merged with ```@chaintip fixes #114``` in the creation comment.

omersiar, 2018-05-15:
Clear Log butonu calisiyor mu? Chrome f12 gelistirici araclarinda network bolumu var, orada ws sekmesi var veri gelip gidiyor mu?

araziforum, 2018-05-15:
sadece sayfayı yenile dediğimde şifre soruyor o sırada veri akışı oluyor dediğiniz yerde ancak menüler arasındaki geçişte herhangi bir değişiklik olmuyor network bölümünde. access log bölümüne clear yaptığımda da bişey olmadı. sadece f5 yapınca orada akış oluyor

araziforum, 2018-06-21:
802.1x eap Wifi bağlantısı yapabilmek için Wifi bağlantı ayarlarına kullanıcı adı ve şifre girecek şekilde bir bölüm yapılabilir mi?

m00st, 2018-06-27:
Here is a very nice implementation using a simple 5v buzzer and one or more ws2812 leds: https://www.youtube.com/watch?v=BvjtoR0eq_M

Similar functionality would be really aweseome.

closed 2018-07-01 by omersiar