omersiar/esp-rfid #116: Time zone doesn't works

quikote, 2018-04-29:
It doesn't matter which time zone I select. The device time is always 1 hour behind the real time in my country. The sync brower time to device doesn't fix the problem and neither the NTP server from my country.

omersiar, 2018-06-09:
@quikote Hello,

Is this can be bacause of DST (Daylight Saving Time)? It is hard to know every countries' DST policy, for example I do not even know my country's current DST policy.

Basically you need to set the time difference between your local time and UTC time.  

For example we were on GMT +2 country list but now, we did not rolled back to DST time so we become a GMT +3 country. (this is pure madness)

Also if this is not the case i need to know your settings.

omersiar, 2018-07-01:
Is this still an issue?

omersiar, 2018-07-31:
Standart getTimezoneOffset() function returns invalid offset for Turkey on Chrome, Firefox does not have this issue, so this is a chromium bug. Reported to Chromium Devs.

closed 2018-07-31 by omersiar