omersiar/esp-rfid #117: Websocket or MQTT with Nodered?

Svellem, 2018-05-02:
Has anyone got a working setup with either websocket or MQTT to Nodered? My aim is to send RFID id's to Nodered and use that for various automations.

I am having trouble connecting the Nodered WS client to the websocket server on the ESP-RFID. Testing via a WS client in Chrome and connection and messages are fine. 

MQTT is working to some extent. Once a card is read, it will only send another message to the MQTT broker when another RFID card is read. This means you have to continuely swap which card you use in order for an MQTT message to be generated.

I am currently using 0.74 firmware.

Great work Ömer :) @omersiar

magic-tune, 2018-05-03:
@Svellem
I have a setup with mqtt up and running for 3 weeks now. Can confirm that during my testing the same card works only once with mqtt. I solved it by resending a timestap and username to the same topic by my ha. This way i can also display the last entered person. I dont use Node Red for now, but played a while ago with. I am sure it is  possible.
I think this is not a esp-rfid issue. Maybe the broker config needs some more input for resending same messages?

Svellem, 2018-05-07:
@omersiar & @magic-tune 
Managed to resolve this issue by reapplying mqtt subscription topic on the Home automation/Node red side. Now each event triggers an mqtt message which again triggers subsequent automations. I did not have any luck with websocket within Node red, but I consider it a Node red issue. Thanks for your input :)

closed 2018-05-07 by Svellem