omersiar/esp-rfid #118: mqtt topics

vr6syncro, 2018-05-02:
in the last fw 0.7.4 the firmware after update was shown as v0.7.3 

and in the mqtt menue i think there should be 2 options. uid and label

both were send to server but on same topic is it possible to ad a topic for uid and one for label?

quikote, 2018-05-02:
I agree. The topic could be the one we can tye in the MQTT setup page, but it would be a good idea that internally the username and UID were sent in different sub-topics. I think it is not necessary to personalize those sub-topics, because for example “username” and “UID” would be self explained.
 
For example, if I choose as MQTT topic “main_door”, the response sent to the broker in every access should be “main_door/username/[USERNAME]” and “main_door/uid/[UID]

Another sub-topic sent could be  “main_door/LWT” for the Last and Will Testament ;-))))

omersiar, 2018-06-09:
I need to understand users' need.  Thank you for your feedbacks, i think hostname can be used for main topic.  If someone point me out the correct approach that would be much appreciated.

vr6syncro, 2018-07-06:
Something like Jens posted would be nice.   What help do u need?

omersiar, 2018-07-06:
I did not know we could also push some json data, good to know.

Some example would be nice.

jensfr1, 2018-07-06:
The example is coming from a standard tasmota flashed sonoff.

ForSquirel, 2018-07-31:
The project is great and I have no doubt I'll implement it in my automation setup, but a few things need 'updating'.

It would be nice if topic was pulled from the MQTT setup page, that way LWT could be implemented as well as connection message. Removing the 'test' topic and using the publish topic would be better as well, especially for those that have extensive MQTT setups in home. Seeing a 'lol' message going across the screen is a bit confusing.

omersiar, 2018-08-03:
@ForSquirel sorry for the inconvenience about test message, i forgot it in production.

MQTT will be more matured on next release.

omersiar, 2018-08-03:
Closing this issue for now, let's track them on #125

closed 2018-08-03 by omersiar