omersiar/esp-rfid #120: Mqtt message??

9acca9, 2018-06-06:
Hi to all.
I just config mosquitto on Ubuntu 16.04. and config esp-rfid to direct the message to that server.....
And i see in mosquitto server that there is connection between the machines, the esp-rfid is sending the uid and name of the users. But, how can i, for example "open the door", from mosquitto server. How can i send a message to esp-rfid.

Thanks to all!!!

(i dont speak english)

omersiar, 2018-06-06:
Hello,

You can not do that with mqtt, you need to trigger websocket message for example with node red

9acca9, 2018-06-06:
Hi, and thanks.
Sorry my ignorance, but i already need mosquitto???
it will be something like esp-rfid-----------mosquitto server------NodeRed???
And what would be the "messages" to the "topic"???
thanks-

9acca9, 2018-06-06:
Hi, and with MQTTBOX?? without NodeRed??

What would be the message to "open the door"? with mqttbox. I need to change the source code??
thanks

nbaglietto, 2018-06-07:
Hello.
Can somebody give me a hand to make a node red "app", just a idea, to control the esp-rfid (i use it to open a door office). So, just want a button to open that door. I read in other post that i need to config websocket.

Thanks for your time.

Attach verbose and command result and  picture with diagram link (G drive): https://drive.google.com/open?id=1w0Q79JWsZ_oI0xd3EO9SB4Xvfd_qyVth

 mosquitto_sub -h localhost -t "esp-rfid/#" -v
esp-rfid xxxxx
esp-rfid pepe cantero

mosquitto –v

1528396627: mosquitto version 1.4.8 (build date Thu, 01 Mar 2018 09:34:49 -0500) starting
1528396627: Using default config.
1528396627: Opening ipv4 listen socket on port 1883.
1528396627: Opening ipv6 listen socket on port 1883.
1528396657: New connection from 192.168.4.4 on port 1883.
1528396657: New client connected from 192.168.4.4 as esp82660eac13 (c1, k15, u'').
1528396657: Sending CONNACK to esp82660eac13 (0, 0)
1528396657: Received PUBLISH from esp82660eac13 (d0, q1, r1, m1, 'test/lol', ... (6 bytes))
1528396657: Sending PUBACK to esp82660eac13 (Mid: 1)
1528396668: Received PINGREQ from esp82660eac13
1528396668: Sending PINGRESP to esp82660eac13
1528396679: Received PINGREQ from esp82660eac13
1528396679: Sending PINGRESP to esp82660eac13
1528396689: Received PINGREQ from esp82660eac13
1528396689: Sending PINGRESP to esp82660eac13
1528396700: Received PINGREQ from esp82660eac13
1528396700: Sending PINGRESP to esp82660eac13
1528396711: Received PINGREQ from esp82660eac13
1528396711: Sending PINGRESP to esp82660eac13
1528396721: Received PINGREQ from esp82660eac13
1528396721: Sending PINGRESP to esp82660eac13
1528396727: New connection from ::1 on port 1883.
1528396727: New client connected from ::1 as mosqsub|9654-mosquito (c1, k60).
1528396727: Sending CONNACK to mosqsub|9654-mosquito (0, 0)
1528396727: Received SUBSCRIBE from mosqsub|9654-mosquito
1528396727:     test/# (QoS 0)
1528396727: mosqsub|9654-mosquito 0 test/#
1528396727: Sending SUBACK to mosqsub|9654-mosquito
1528396727: Sending PUBLISH to mosqsub|9654-mosquito (d0, q0, r1, m0, 'test/lol', ... (6 bytes))
1528396730: Socket error on client mosqsub|9654-mosquito, disconnecting.
1528396732: Received PINGREQ from esp82660eac13
1528396732: Sending PINGRESP to esp82660eac13
1528396737: New connection from ::1 on port 1883.
1528396737: New client connected from ::1 as mosqsub|9655-mosquito (c1, k60).
1528396737: Sending CONNACK to mosqsub|9655-mosquito (0, 0)
1528396737: Received SUBSCRIBE from mosqsub|9655-mosquito
1528396737:     esp-rfid/# (QoS 0)
1528396737: mosqsub|9655-mosquito 0 esp-rfid/#
1528396737: Sending SUBACK to mosqsub|9655-mosquito
1528396743: Received PINGREQ from esp82660eac13
1528396743: Sending PINGRESP to esp82660eac13
1528396747: Received PUBLISH from esp82660eac13 (d0, q0, r1, m0, 'esp-rfid', ... (8 bytes))
1528396747: Sending PUBLISH to mosqsub|9655-mosquito (d0, q0, r0, m0, 'esp-rfid', ... (8 bytes))
1528396747: Received PUBLISH from esp82660eac13 (d0, q0, r1, m0, 'esp-rfid', ... (16 bytes))
1528396747: Sending PUBLISH to mosqsub|9655-mosquito (d0, q0, r0, m0, 'esp-rfid', ... (16 bytes))
1528396753: Received PINGREQ from esp82660eac13
1528396753: Sending PINGRESP to esp82660eac13

(i dont speak english)

omersiar, 2018-07-01:
I do not know if Node-Red supports protocol upgrade from HTTP to WebSocket. You will definetly need that to send websocket messages to esp-rfid, this is because the Authentication must be done with HTTP protocol and then it needs to switch to WebSocket protocol.

closed 2018-07-01 by omersiar