omersiar/esp-rfid #123: Problem Wifi maybe router ?

Chopin85, 2018-07-02:
Hi everyone it's been a long time I follow this project I tried many versions without success.
When I connect to an access point such as my iphone works, I can have a valid ip so in the serial monitor I write the ip associated with the nodecmu and at the end DONE instead if I connect to my home router the serial minitor writes the ip given to the nodecmu but does not write DONE so I can not connect with continuous restarts.
I tested however that if I impose a static ip on nodecmu it works. Why is this the case?
Another issue the nodemcu lose the connection established with the wifi the only way to reconnect is a reset from the button of the nodemcu ?

Thanks

I wanted to congratulate you on this beautiful software.

Chopin85, 2018-07-02:
I just tested the latest version is everything works thanks.  The solution is 

https://github.com/omersiar/esp-rfid/commit/a1391aa1f9d06910497822c71d0e532726a70a7f

It now works perfectly.  
When I disconnect from Wi-Fi, will I reconnect ?

omersiar, 2018-07-04:
I should prepare a new version from it.  Thank you for your feedback.

closed 2018-07-31 by omersiar