omersiar/esp-rfid #124: current platform triggers restart for wiegand readers

nardev, 2018-07-05:
Current plarform set in platformio.ini triggers ESP8266 restart. However, i didn't notice any errors if we use
platform = espressif8266
with
build_flags = -Wl,-Tesp8266.flash.4m.ld

I was not abler to check for other readers.

omersiar, 2018-07-05:
2mb firmware, 2 mb spiffs allows us to make ota updates bigger than 550kb firmwares,  i think we can make the firmware small enough to fit in first 500 kb, but this requires multiple optimizations.

It must be a bug in the staging version of esp8266 core, it will be mature eventually.

nardev, 2018-07-05:
Hey @omersiar, what do you think about making separate builds for different reader types? I assume that once you determine the reader type you would not change it for a while. And if you are developer, all readers version is ok for you, OTA would not be something importand if you constantly have the ESP on your table.

omersiar, 2018-08-08:
We are working on different solution for OTA update.
I dropped using of staging version of Core

closed 2018-08-09 by omersiar