omersiar/esp-rfid #125: MQTT Format

jensfr1, 2018-07-05:
The MQTT format is not really user friendly. It would be great if the output would be like this

topic/username/STATE {.....
e.g.
tele/sonoff-keller/STATE {"Time":"2018-07-05T20:47:38","Uptime":"17717T11:47:37","Vcc":3.459,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wlan","RSSI":100,"APMac":"1111"}}

Thanks, Jens

cyclemat, 2018-07-20:
better is you make it in the JSON string this is the  the only way you can work with the data like
the system must work this way

RFIDTOPICNAME/ TIME:, USER: , CARDID:as45464, ACESS:YES/NO, RELAIS ACTIVE:YES/NO, WIFISIGNAL:100,

omersiar, 2018-08-03:
We are making it JSON formatted text on next release.

omersiar, 2018-08-03:
- [ ] Fix retained publish
- [ ] Implement LWT
- [ ] Return JSON formatted text to MQTT Broker

omersiar, 2018-08-08:
We won't have the LWT, at least not in a way that MQTT protocol specifies but now we have heartbeat and boot messages.

closed 2018-08-09 by omersiar

rijulmanch, 2019-07-10:
@omersiar  what's the issue with LWT?