omersiar/esp-rfid #126: Update not working

jensfr1, 2018-07-06:
I tried the update via the web interface but it is not working. It always shows version 0.73. I tried updating to 0.74 and 0.75.

Chopin85, 2018-07-06:
Hello your update does not work because you use flash.bat with firmware.bin that is inside the bin folder.
Run erase.bat in the bin folder and after flash.bat replacing firmware.bin with the attached one.

[firmware.zip](https://github.com/omersiar/esp-rfid/files/2170972/firmware.zip)

jensfr1, 2018-07-06:
I was using the update from the web ui... with that file it is working from webui.

Jafterdark, 2018-07-11:
well I didn't want to start a new issue but it's more of a question is there anyway if you have multiple doors and you have these setup that you can configure it under one instead of each one that you have

PrathikGopal, 2018-07-14:
I tested the same on webui, after i select the file it reboots and nothing happens, Same results as @jensfr1 , thanks again for your help in advance.

omersiar, 2018-07-31:
Always use https://github.com/omersiar/esp-rfid/releases binaries, but first wait for the 0.7.7 release which will fix some bugs with versioning, then I will abandon the pre-built binary concept which is confusing.

closed 2018-08-09 by omersiar