omersiar/esp-rfid #131: Problem with rc522

Chopin85, 2018-07-31:
Hello, i have a problem rfid RC522, when I turn it off and on again, it works  but when it is active for more than one day, the reader not detects any tags but the web interface works, so I can trigger relay with interface web.
Did someone else have this problem ? Say it's better to solder the pin on nodemcu esp8266 to RC522 ?
Installed firmware on esp8266: v0.7.6.

Thanks.

omersiar, 2018-07-31:
in this case i would recommend changing the mfrc522 module, it may be faulty, also proper powering is essential for both esp and rc522.

Chopin85, 2018-07-31:
Thanks for your quick response, the power supply is taken from the nodemcu. 
The strange thing that for a time works.  Anyway I have another one at home, I will try to change it.
 I'll update you.

Thanks.

omersiar, 2018-07-31:
First things first, we need to eliminate the hardware problem (if any) before looking into software.

closed 2018-08-09 by omersiar