omersiar/esp-rfid #132: Wiegand keypad support?

koffienl, 2018-08-01:
I stumbled upon this project and it looks very promising. In some spare time I'n building something similar but it's not as extended as this project.
I'm using a wiegand RFID with keypad reader in my project. Looking at the demo and source code it looks like there is no keypad support yet? Is this something that will be added in the future? 
Looking through the issues I also saw some questions about support for buzzer, would be a very nice to have to enable.
With buzzer support you could send a command to the ESP to start buzzing so the user is notified to enter a code / tag. Also after accepting a code it could start buzzing to notify the user the system is armed.

Some code snippets I use for keypad support:"
````
  int timeout = 10000;
  long KeyTimer = 0;
  int keylength = 4;

  // Keep an eye on timeout waiting for keypress after starting with a *
  // Clear code and timer when timeout is reached
  if (KeyTimer > 0 && millis() - KeyTimer >= timeout)
  {
    Serial.println("Keycode timeout");
    CurrentInput = "";
  }

  // When * is pressed start keytimer to capture code
  if (String(wg.getWiegandType()) == "4" && String(wg.getCode(), HEX) == "1b")
  {
    Serial.println("Start capture keycode . . .");
    KeyTimer = millis();
  }

  // When key is pressed AND timer is running AND CurrentInput < keylength AND key is not * or #
  // Then add key to the current input
  if (String(wg.getWiegandType()) == "4" && KeyTimer > 0 && CurrentInput.length() < keylength && String(wg.getCode(), HEX) != "d"  && String(wg.getCode(), HEX) != "1b")
  {
    CurrentInput = CurrentInput + String(wg.getCode());
    KeyTimer = millis();
  }

  // When captured code equals predefined keylength, stop timer and check given code
  if (CurrentInput.length() == keylength)
  {
    Serial.println("Stop capture keycode . . .");
    Serial.println(CurrentInput);
  }
````

This snippet has a hardcoded length for keypad input, but you could change it to not only start with a ````*```` but also end with a ````#```` . Further this snippet uses a timeout to prevent the code to wait endless for the user to finish the pincode.

koffienl, 2018-08-01:
OK, for the fun of it I did a quick add of code to the ```main.cpp``` and it seems to be working. User keypad code looks like it can be quit long (tested with 123456789 as pin). To start keypad you press ```*````` and to finish you press ````#```` everything between will be seen as the pin code.
the PICC type of the input will be set as ````PIN```` 

Add line 75 add
```` 
// Variables for whole scope
int timeout = 10000;
long KeyTimer = 0;
String CurrentInput = "";
```` 

Add line 500 replace this:
```` 
            Serial.print(F("[ INFO ] PICC's UID: "));
            Serial.println(wg.getCode());
            uid = String(wg.getCode(), HEX);
            type = String(wg.getWiegandType(), HEX);
            cooldown = millis() + 2000;
```` 

with this:
````
          // Detect Wiegand26 and Wiegand34 keyfob
          if (String(wg.getWiegandType()) == "26" || String(wg.getWiegandType()) == "34")
          {
            Serial.print(F("[ INFO ] PICC's UID: "));
            Serial.println(wg.getCode());
            uid = String(wg.getCode(), HEX);
            type = String(wg.getWiegandType(), HEX);
            cooldown = millis() + 2000;
          }

          // keypad stuff
          // When * is pressed start keytimer to capture code
          if (String(wg.getWiegandType()) == "4" && String(wg.getCode(), HEX) == "1b")
          {
            Serial.println("Start capture keycode . . .");
            CurrentInput = "";
            KeyTimer = millis();
          }

          // When key is pressed AND timer is running AND key is not * or #
          // Then add key to the current input
          if (String(wg.getWiegandType()) == "4" && KeyTimer > 0 && String(wg.getCode(), HEX) != "d"  && String(wg.getCode(), HEX) != "1b")
          {
            CurrentInput = CurrentInput + String(wg.getCode());
            KeyTimer = millis();
          }

          // When # is pressed stop keytimer to capture code
          if (String(wg.getWiegandType()) == "4" && KeyTimer > 0 && String(wg.getCode(), HEX) == "d")
          {
            Serial.println("Stop capture keycode . . .");

            Serial.print(F("[ INFO ] PICC's UID: "));
            Serial.println(CurrentInput);
            //Serial.println(wg.getCode());
            uid = CurrentInput;
            type = "PIN";
            CurrentInput = "";
            KeyTimer = 0;
            cooldown = millis() + 2000;
````

Ravens28, 2018-08-02:
Thank you for this add!
I was also looking for this enhancement.
It would be really interesting to see it in the initial code.

Regards

omersiar, 2018-08-03:
a pull request would be nice. @koffienl If you can test these changes in your setup and come with conclusion that is stable enough i can make it to include on mainstream code base.

koffienl, 2018-08-03:
@omersiar I'll try in a few days. This was just quick copy / paste of some code I had. Have to take a better look if it's all OK.

kodibrain, 2018-08-04:
hi @koffienl thanks for this add. I have issues when compiling.

`src\main.cpp: In function 'void rfidloop()':
src\main.cpp:715:59: error: a function-definition is not allowed here before '{' token
void ICACHE_FLASH_ATTR setupWiegandReader(int d0, int d1) {
^
src\main.cpp:720:69: error: a function-definition is not allowed here before '{' token
void ICACHE_FLASH_ATTR setupMFRC522Reader(int rfidss, int rfidgain) {
^
src\main.cpp:736:53: error: a function-definition is not allowed here before '{' token
void ICACHE_FLASH_ATTR setupPN532Reader(int rfidss) {
^
src\main.cpp:1385:1: error: expected '}' at end of input
}
^
*** [.pioenvs\nodemcu\src\main.cpp.o] Error 1`

kodibrain, 2018-08-04:
hi @koffienl thanks for this add. I have issues when compiling.
[image]

kodibrain, 2018-08-04:
ah ok it is missing } at the end of replace add line 500 :)

nardev, 2018-08-04:
Nice to see this, i had my own code, didn't try to integrate it with repo but it's great to see it.

Also, the buzzer and LED control is important to be added into a repo. Wiegand readers often have those two contrary to other simple RFID boards.

nardev, 2018-08-04:
For anyone interested in ESP-RFID board with relay, for Wiegand readers, here is more about it.

https://github.com/omersiar/esp-rfid/issues/133#issuecomment-410448552

p.s. check the discount code.

nardev, 2018-08-04:
@koffienl contact me please i want to send you this board if you want.

koffienl, 2018-08-05:
Been offline couple of days due to holiday :)
@nardev Not sure if your wiegand reader is the same, but mine also has option for blinking the LED and beeping the buzzer. Unfortunately the 2 wires needs to be connected to the ground so I'm afraid you would need a relay in between for this to connect and use, or is that already in the board?
Interested in the board but my solder skills are not that great. I'll contact you tomorrow.

For buzzing : personally I need quite a different approach for buzzing that would be to much to incorporate in the ESP-RFID project. I used several type of beeping methods in my project, all based on a simple piezo buzzer and PWM script:

````
#include <Ticker.h>
Ticker BeeperSlow_ticker;
Ticker BeeperFast_ticker;

// Buzzer
#define NOTE_C7  2093
#define NOTE_E7  2637
#define NOTE_G7  3136
#define NOTE_E6  1319
#define NOTE_G6  1568
#define NOTE_D7  2349
#define BuzzerGPIO 5

int BeeperSlow_timeout = 10000;
long BeeperSlow_timer = 0;
int BeeperFast_timeout = 5000;
long BeeperFast_timer = 0;
String CurrentBeeper;

void BeeperSlow()
{
  if (millis() - BeeperSlow_timer > BeeperSlow_timeout)
  {
    BeeperSlow_ticker.detach();
    BeeperFast_timer = millis();
    CurrentBeeper = "fast";
    BeeperFast_ticker.attach(0.650, BeeperFast);
  }

  if (StopBeep == 1)
  {
    Serial.println("Self detach");
    BeeperSlow_ticker.detach();
    StopBeep = 0;
  }
  else
  {
    long delayValue = 1000000 / NOTE_C7 / 2; // calculate the delay value between transitions
    long numCycles = NOTE_C7 * 300 / 1000; // calculate the number of cycles for proper timing
    for (long i = 0; i < numCycles; i++)  // for the calculated length of time...
    {
      digitalWrite(BuzzerGPIO, HIGH); // write the buzzer pin high to push out the diaphram
      delayMicroseconds(delayValue); // wait for the calculated delay value
      digitalWrite(BuzzerGPIO, LOW); // write the buzzer pin low to pull back the diaphram
      delayMicroseconds(delayValue); // wait again or the calculated delay value
    }
    Serial.println("SLOW piep!");
    delay(100);
  }
}

void BeeperFast()
{
  if (millis() - BeeperFast_timer > BeeperFast_timeout)
  {
    BeeperFast_ticker.detach();
    //BeeperFast_timer = millis();
    //CurrentBeeper = "fast";
    //BeeperFast_ticker.attach(1, BeeperFast);
  }

  if (StopBeep == 1)
  {
    Serial.println("Self detach");
    BeeperFast_ticker.detach();
    StopBeep = 0;
  }
  else
  {
    long delayValue = 1000000 / NOTE_C7 / 2; // calculate the delay value between transitions
    long numCycles = NOTE_C7 * 300 / 1000; // calculate the number of cycles for proper timing
    for (long i = 0; i < numCycles; i++)  // for the calculated length of time...
    {
      digitalWrite(BuzzerGPIO, HIGH); // write the buzzer pin high to push out the diaphram
      delayMicroseconds(delayValue); // wait for the calculated delay value
      digitalWrite(BuzzerGPIO, LOW); // write the buzzer pin low to pull back the diaphram
      delayMicroseconds(delayValue); // wait again or the calculated delay value
    }
    Serial.println("FAST piep!");
    delay(100);
  }
}

void buzz(int targetPin, long frequency, long length, int repeats) {
  for (int i = 0; i < repeats; ++i)
  {
    long delayValue = 1000000 / frequency / 2; // calculate the delay value between transitions
    //// 1 second's worth of microseconds, divided by the frequency, then split in half since
    //// there are two phases to each cycle
    long numCycles = frequency * length / 1000; // calculate the number of cycles for proper timing
    //// multiply frequency, which is really cycles per second, by the number of seconds to
    //// get the total number of cycles to produce
    for (long i = 0; i < numCycles; i++) { // for the calculated length of time...
      digitalWrite(targetPin, HIGH); // write the buzzer pin high to push out the diaphram
      delayMicroseconds(delayValue); // wait for the calculated delay value
      digitalWrite(targetPin, LOW); // write the buzzer pin low to pull back the diaphram
      delayMicroseconds(delayValue); // wait again or the calculated delay value
    }
    delay(100);
  }
}

void BeepOK()
{
  buzz(BuzzerGPIO, NOTE_C7, 100, 3);
  buzz(BuzzerGPIO, NOTE_C7, 300, 1);
}

void BeepWrong()
{
  buzz(BuzzerGPIO, NOTE_C7, 300, 2);
}
````

This would give the following options:
Call the ````BeepOK()```` function to inform the user a code is accepted
Call the ````BeepWrong()```` function to inform the user a code (or something else) is wrong
Call the ````BeeperSlow()```` function that will slowly start to beep and after a period of time increase the speed of beeping. This 'loop' can be stopped by setting ````StopBeep = 1````
This kind of beeping behavior is more interesting if you use reader for a DIY alarm. When the system is armed and there is movement the beep loop will be initiated to inform the user a code/card is needed. Upon a correct code the beep loop is stopped and a ````BeepOK()```` will be played.
But again this is probably way off scope of the project, so I guess I will move only the beeping part of my code to a separate ESP8266 that would only have one goal : play some bleeps based on incoming MQTT messages.

I'll be happy to discuss thoughts and ideas for a DIY alarm in a separate topic if there is interest by others.

nardev, 2018-08-05:
@koffienl you would get it complete, just to hook the reader.
Please, can you send me photo of your reader?

The readers i have been using already have a beep on every reed, but, after that initial reading beep, you can add play something you want and play with LED.

koffienl, 2018-08-06:
Very much interested in testing. I'll send you an email.

koffienl, 2018-08-06:
Mail send, bu having issues with my mail so I can't add attachments. I'll fix that tomorrow so I can send you the promised pictures.

nardev, 2018-08-06:
@koffienl hey, I got the email,  you can also upload image here on github's comment.

koffienl, 2018-08-15:
Added the code to the 0.8.0 version, works nice. Hope to do a longer period test soon

koffienl, 2018-09-03:
I'm running a official board from @nardev with editted 0.8.0 to use the keypad. So far I have zero issues.
Not using the cooldown after each keypress is intentional, because mine (and most other wiegand keypad readers) have build-in cooldown. When I'm really fast a keypress might be lost, but that's not due to the code. When I press a number twice very fast the reader doesn't give feedback with LED and beep, so it's a timing 'issue' on the reader and not in the code or board.

omersiar, 2018-09-26:
@koffienl Hello,

Are there any updates on this?

koffienl, 2018-09-26:
The only update I can give is that it works without any issues at all. Have to build my second reader (same specs, different manufacturer/look) but a 5 minute test with that one was also succes.
I guess it can be incorperated into the code ?

omersiar, 2018-09-26:
Yeah i would like to do that.

If this is the latest version of your code let me know, then i will copy it to my branch.
https://github.com/koffienl/esp-rfid/commit/0fa2b662ab84111c7ac831d0423f03e57ca2da20

nardev, 2018-09-26:
Than, it would be useful to add pin field in users lists next to card number. What others think?

omersiar, 2018-09-26:
@nardev we need to test how long does it take to find given pin number in users files. I do not think it can scale with large numbers of users.

This will require to look into every record (parsing json and comparing).

Also how one can choose pin ? What happens when two user wants to have same pin? We can avoid that with additional check but at what cost.

nardev, 2018-09-26:
I had this case, in fact i have implemented this but in different environment.. well the procedure is straight forward.. i'll try to make an example in separate branch...

kessack, 2018-12-03:
This will be really useful when working. I have a NodeMCU ESP8266 12E. I can flash sucessfuly with these binaries:   1) 0.9.0     2) 0.8.3 and    3) 0.8.1.       I can connect to each wifi hotspot and then see the page of each with 192.168.4.1. However only the last of these binaries will save the new client settings and then let me connect through my local modem router. MAy be I need to compile with platformio and not use the provided compiled binaries?

nardev, 2018-12-04:
My suggestion, try to build from source. It's not hard`

omersiar, 2018-12-04:
@kessack The only version currently supported is v0.8.1, sorry for the confusing releases lastly. I will remove confusing release now.

nardev, 2018-12-26:
@omersiar i'm sorry i didn't trace the reason, why it was not included in 0.9?

I'm just working to merge it with my branch, with some other buzzer and LED options.... but "my way"

Is there any suggestion suggestion you can offer me? Any particular guideline what to add?

nardev, 2018-12-26:
@koffienl is it ok if i use part of your code for buzzer? It will be part of new signaling functionalities i'm making for buzzer and for LED signalization.

koffienl, 2018-12-29:
> @koffienl is it ok if i use part of your code for buzzer? It will be part of new signaling functionalities i'm making for buzzer and for LED signalization.

Of course, no problem. It's just a simple buzzer :)

svrooij, 2019-01-12:
@omersiar what would need to happen to have this implemented? I got already got your board in the mail and I would really like it with the support for pin’s.
Do you need some help with that?

omersiar, 2019-01-12:
I believe @koffienl's code is matured by now, i do not have the hardware myself, so i am waiting for a pull request.

Maybe @nardev will make it happen when he has time.

@nardev there was a websocket bug with the 0.9.x where the settings were not be saved, @Pako2 solved it recently.

And current dev branch can now be worked on it.

svrooij, 2019-01-12:
Want me to donate money so you can buy this keypad? 
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.nl%2Fulk%2Fitm%2F232384491546

omersiar, 2019-01-12:
Thank you for the offer, but i can not accept it. You can put a bounty on this issue though, it may be picked up by willing coder.

nardev, 2019-01-14:
> I just posted a bounty for this issue. Hopefully this will get you guys motivated to fix this faster.
> https://www.bountysource.com/issues/61614375-wiegand-keypad-support

I did it but didn't try to push it. I had huge issues back at home so.. :(

I hope today..

nardev, 2019-01-29:
Here is the thing.

I spend quite some time about this issue. And at the end, it was not possible to just add this easily.

The code that i made has some fundamental changes on esp-rfid and i would like to hear your thoughts.

Here is my explanation:

- So far, the uid was card number because it was only relevant index, there was no need to handle duplicate checks or have any mechanism of validation

- I had to change that approach and uid now in approach is actually timestamp or randomized alphanumeric string of 12 chars.

- That way, i believe we could be pretty safe even when in future if have multiple devices interconnected and exchanging the cards and access lists. Because of that change, now in user file i have more data. I have "card and "pin" and uid is re purposed as i explained.

- Another problem is import/export which i'm yet to complete. And for sure it will require procedure for importing old user config files.

- Also, what has changed is adding and editing new users. In order to do that, every time user is added or created, the card number and pin must be checked in order to confirm that it doesn't exist already.

- Also, in order to validate pin/card, now there is "cards" and "pins" json index that is always generated on device start OR it could be switched to "/users/cards/[card_number]" and "/users/pins/[user_pins]" index system which actually consists files which would hold user IDs "uid" numbers.

So the validation would go like:
`Enter Pin or Swipe Card => Check Pin/Card in Pin or card Index files or Json boject => if exists, get the UID and check if "/users/[uid_number]" file exists => if it does, open it and check permissions. => if ok, open the doors.`

Right now  it works like this:
`we directly just validate if card number exists as file name in "/P/[uid]" and then => check permissions and see if it is ok to open doors.`

**So, do you want me to push this?**

nardev, 2019-01-29:
I must admit that the puzzle is very interesting approach. I haven't seen it implemented.

You are right about the "complexity" in the base solution that i mentioned. But i was guided by some industry expected standard solution. It's pretty much what any developer would expect it to be.

mcd1992, 2019-02-26:
Was contemplating opening another issue but this one seems relevant.

As a feature request it might be nice to have a 'two-factor auth' plugin system. Let people write plugins for something like keypad 2fa or even just a telegram message for 2fa.

omersiar, 2019-02-26:
@mcd1992 nice idea, the only problem i can imagine is blocking nature of keypads (i am not sure btw)

stephanflug, 2019-03-05:
Hallo bin hier neu und habe eine Frage.Habe folgendes Gerät im Einsatz https://www.amazon.de/gp/product/B07KZX57JV/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
die RFID-Karte werden richtig angezeigt nur beim eingeben eines Zugangscode über der Tastatur kommt immer nur die erste Zahl und nicht mehr.

nardev, 2019-03-05:
@stephanflug because keypad support is stil pending. I have the code for it but i have to finish testings. in 1.1. it will be available.

Netoperz, 2019-03-30:
I do see some potential problems with this implementation as far as i can see from proposals.

1. Some keypads that are standalone controllers with terminal mode as an option, use * to enter to the programming sequence. So when user is pressing * the keypad is waiting for the mastercode of that particular hardware for entering setup. I have got 8 or 10 keypads and they all do the same.

More secure approach is to check for input,and put the input to the buffer, counting the buffer content.
In my co i'm forming the bufffer, and filling with -1 (impossible input from keypad) than i just write the keypad input. and i'm checking if the particular place in buffer is different than -1, for example for 5 digit pin if buffer4 >-1, than the buffer goes to char array and is forwarded to verification if it exist in the database.

As for the "time for input"  and  "clearing the incomplete input" i'm using the systicks to count the loops, and i'm checking if buffer 0> -1 if it is, and counter limit is reached, and the pin digit input number is not reached the buffer is cleared, exactly filled with -1.

This solution works perfect over two years on 16 doors in publicly used building.

here is the code of the wiegand task. Also You need to initialize the buffer and so on...

`
unsigned int WiegandInputCouter = 0;
unsigned int WiegandInputCouterLimit = 150;

void WiegandWorkerTask(void *parameter)
{
  while (true)
  {
    if (wg.available())
    {
      runInputConfirm();
      WiegandInputCouter = 0;
      Serial.print("WG HEX = ");
      Serial.print(wg.getCode(), HEX);
      Serial.print(", DEC. = ");
      Serial.print(wg.getCode());
      Serial.print(", WG TYPE ");
      Serial.println(wg.getWiegandType());
      unsigned long currentCodeProcessing = wg.getCode();
      if (isCardCode(currentCodeProcessing))
      {
        processingCardCode(currentCodeProcessing);
        countCode = 0;
        clearBuffer();
      }
      else
      {
        countCode++;
        if (countCode < 6)
        {
          codeBuffer[countCode - 1] = wg.getCode();
          LCD__keypad_input_counter();
          for (int i = 0; i < 5; i++)
          {
            clearBuffertIfHash();
            Serial.println(codeBuffer[i]);
          }
          if (codeBuffer[4] > -1)
          {
            processingCodeViaBase();
            countCode = 0;
            delay(8); // was 1000
          }
        }
      }
    }
    else
    {
      if (codeBuffer[0] > -1)
      {
        WiegandInputCouter++;
        if (WiegandInputCouter == WiegandInputCouterLimit)
        {
          WiegandInputCouter = 0;
          countCode = 0;
          clearBuffer();
          Serial.println(
              "CLEAR MEM. BY W.I.C.");
        }
      }
    }
    delay(5);
  }
}

====
clear the buffer manualy using #
void clearBuffertIfHash()
{
  for (int i = 0; i < 5; i++)
  {
    if (codeBuffer[i] == 13)
    {
      WiegandInputCouter = 0;
      countCode = 0;
      memset(codeBuffer, -1, sizeof(codeBuffer));
      LCD_closed_status_screen();
      runBuzzerClear();
//      runLedClear();
      Serial.println("CLEAR MEM. #");
    }
  }
}

`

I'm not sure that i will help here, but this approach is more user friendly than need to press special keys, and you may define the pin code length modifying the code to be relative, it's just some code i had around and i tough that it may help as an idea.

nardev, 2019-03-30:
The keypad is not problem at all, the issue is once you make it, you have to implement pin as accessing way.

However, then, you would have parallel pin and card as ide, than there is an issue because you would have to maintain two different index lists and whole sets of functions to maintain unique pins and cards and combinations. and so on...

Netoperz, 2019-03-30:
@nardev 
In my model, both pincodes and cards are using the same index list, in backend panel You can input card number on pin in the same field.
Mifare Card Serial Number is the unique identifier defined in ISO 14443-3A. There are 3 types of UID defined in the standard - single ( 4Byte ), Double ( 7 bytes ) and Triple ( 10 bytes ). Only older versions of the Mifare card, the UID was 4 bytes but now have migrated to 7 bytes.
So in 99% of situations You will have not less than 7 bytes for card.
User pincodes are most likely shorter. I do not know a place where users use 7bytes long pincodes.

That could give a way to filter pincodes vs card numbers (that's what i'm doing in my solution, i've sent you some code to show)

Second solution is as You said to have two acess lists. Reader when You read the card UID sends long value with UID.
When user is inputing pincode it sends one inpyt per keypress, so there is a way to differ the card input and the pin input and veryfy using separate functions on separete lists. Again that's the way i'm doing it.

I'm not the pro developer, i've just learned a bit just to write the functionality, that's all.
Is the pincode functionality availble in dev branch ? or not pushed yet to github ?

omersiar, 2019-04-17:
I think I convinced myself on how to put thorough the keypad on esp-rfid. 

* Independent from Wiegand protocol an **abstraction layer** would cover the supporting the multiple keypad hardware.
* When we done finish implementing hardware abstraction layer, it is only used to consume given pin into HMAC-TOTP to verify the pin, since we already have NTP functionality this would not require a real time clock.
* Devices that are not connected to Internet still will require manual time synchronization or pin codes will get denied if there is time mismatch between User and the esp-rfid.
* With this way any RFC 6238 complaint authenticators can be used like Google's and Microsoft's
* This require users to have their mobile phones to be with them (who doesn't these days?)
* There will be manually generated (each time when all of them are consumed), one time only pin codes (lets generate generously of 10 pin codes) for access without a mobile phone. Since they are ephemeral and consumed each time used they needed to be replaced through WebUI by admin.

akeilox, 2019-04-18:
@omersiar thats a great plan to integrate mobile authenticators like Google and Microsoft.

However what will be the maximum number of records, theoretically, that can be stored in the default flash?

My understanding was that the current build can store up to 1,000 users i.e. cards (I refer that from the hackster io article I read, I might be wrong here).

How such an authenticator pin code integration will affect the max user number in this case ?

omersiar, 2019-04-18:
@akeilox It is based on HMAC-TOTP so it does not need to scale by user number. It just verifies if the given pin is authentic in a time window by cryptographic functions (not by checking given pin in a list of pins).

akeilox, 2019-04-18:
Oh I see what you mean, you get this figured out. 

For clarification what is the max number of users (card ids) that can be supported in current build? I recall there was some sort of test done long time ago but cant seem to find it.

omersiar, 2019-04-18:
https://github.com/esprfid/esp-rfid#scalability

It is not guaranteed, please test yourself and share your findings.

nardev, 2019-04-19:
@akeilox i was able ti test it with 100 cards. However, it was short period of time. It was working fine but i had problems with adding new users, reading logs etc.

akeilox, 2019-04-20:
thanks for the feedback @nardev 
do you think the problems you encountered was related to the flash chip quality or architecture limitations ?

ver2go, 2019-04-29:
First thanks for the work and contributions.  I currently have 1.0.1 working with an Essex KTP keypad.  I can press *<pin># and ESP-RFID will read the input as a single value and assign to a UID.  Working exactly as I want it to. I can't tell from this feature request if this should be working without patch on 1.0.1 or not?  Did I just lucky or it's due to how my Keypad sends the output.

Secondly, because this is working, I went ahead and purchased another wiegand keypad
https://www.amazon.com/gp/product/B01MSZC2AP/
But from debug, this one is not working because of each keypress is being sent individually as a single key being sent to esp-rfid.

My question is what is the expected behavior? If so what is the feature called so it sends as a single value from Essex keypad so I can get another (less expensive) keypad to work with ESP-RFID?

Thanks in advance.

nardev, 2019-04-30:
@ver2go 

If you are using the build with OFFICIALBOARD and if you want to handle keypad inputs, you can do it with this.

Add this in the top, next to other globals in main.cpp

`String uid = "";
String pin = "";
// String cardid = "";
// int inputtype = 0; // rfid read or card read???
String keypadInputStack = ""; // Only for Wiegand Readers!!!
int keypadInputDelay = 2000;
unsigned long previousKeyTime = 0;
unsigned long keypadTimer = 0;
String type = "";`

And add following code in rfid.esp, around line 200, instead of existing OFFICIALBOARD segment.

`#ifdef OFFICIALBOARD
	if (wg.available())
	{
    if (wg.getWiegandType() == 4) {
            if (currentMillis - previousKeyTime >= keypadInputDelay) {
                    keypadInputStack = "";
            }
            previousKeyTime = millis();

#ifdef DEBUG
            Serial.print("Key pressed:"); // put this in debug only!
            Serial.println(wg.getCode()); // put this in debug only!
            Serial.println("==============================="); // put this in debug only!
#endif

            // If key is not ESC (27) or ENT (13) put it on stack!
						// if ((wg.getCode() != 13) && (wg.getCode() != 27))
						if (wg.getCode() < 10) // this way only one compare
            {
              keypadInputStack += String(wg.getCode());

							Serial.print("STACK:"); // put this in debug only!
							Serial.println(keypadInputStack); // put this in debug only!
							return;
            }

						// If the ESC (27) is pressed, then the keypadInputStack is cleared!
            if (wg.getCode() == 27) {
#ifdef DEBUG
                    Serial.print("Entered PIN is cancecled: "); // put this in debug only!
                    Serial.println(keypadInputStack); // put this in debug only!
#endif
                    keypadInputStack = "";
										return;
            }

            // If the key is ENT (13) then the pin is complete and should be checked!
            if (wg.getCode() == 13) {
#ifdef DEBUG
                    Serial.print("Keypad entered PIN: "); // put this in debug only!
                    Serial.println(keypadInputStack); // put this in debug only!
#endif
										if (keypadInputStack == "") {
											return;
										} else {
											uid = keypadInputStack;
										}

                    keypadInputStack = "";
										type = String(wg.getWiegandType(), HEX); // this as it is!!!
										cooldown = millis() + 2000;
            }

    } else if (wg.getWiegandType() == 26 || wg.getWiegandType() == 34) {

#ifdef DEBUG
            Serial.println("Card/Token swiped: "); // -
            Serial.print(wg.getCode()); // -
#endif
						// @nardev: to be used with card+pin version
            // inputtype = 2;

            uid = String(wg.getCode(), DEC); // make this cardid
            // cardid = String(wg.getCode(), DEC); // make this cardid
            type = String(wg.getWiegandType(), HEX); // this as it is!!!
            cooldown = millis() + 2000;

    }
	}
	else
	{
		delay(50);
		return;
	}
#endif`

That way, once you enter the pi, you need to finish it with "ENT" and it will be passed to a controller as if you swiped the card.

However, if more that 2 seconds pass between keys pressed of if you press "ESC" key, the previously entered keys/numbers will be erased.

Contact me via slact https://esp-rfid.slack.com if you get any problems while build.

ver2go, 2019-05-02:
@nardev Thank you. I was able to use your patch.  I'm not sure if I should report my findings here, so apologies in advance.

With my original keypad, I was able to use esp-rfid without patching.  I would press *<pin># and then the keypad outputs to esp-rfid as only output of the wiegand 26-bit format. 
Keypad used:
http://www.keyless.com/ktp-series-thinline-2x6/

Then I tried the Chinese ones from Amazon.  Each key press would send bits so it required @nardev 's patch.  But the values of the key press did not match the number pressed.  I think "1" was like 14 dec.
"*" was 5 dec, "#" was 4 dec, so I had to change the patch. From looking at what I have, the patch, and what I ordered from Amazon, there seems to be 3 variations of how W26 is being handled by each type of keypads.

Any interest in looking into this any further? If so, let me know what I should log.

Thanks again for everyone's contributions.

nardev, 2019-05-02:
@ver2go i think that you mixed wires so you got those crazy values. : )

ver2go, 2019-05-02:
@nardev confirmed.  User error was the cause of the crazy values on the other keypads.  The final tally, I have keypad that will do 26bit burst that doesn't need patch.  A keypad that does 4bit burst, one that does 8bit (W8 according to getWiegandType).  All working now.  Thanks.

peteshoard, 2019-08-29:
Is this feature now in the latest core build?

nardev, 2019-08-29:
@peteshoard no, send me an email, i'll send you the version which has that included and some more mqtt options.

frenchie71, 2019-08-31:
First of all many thanks for our great software and efforts. I have implemented and tested keypad support with my Gelikom XK1 in the stable branch of my fork of your project based on the suggestions here - feel free to use it. Unfortunately I can't submit pull requests that easily because I did a couple of changes  w/r to magic numbers and initially wanted to do HMAC TOTP stuff with Pins over MQTT but abandonned it. [here](https://github.com/frenchie71/esp-rfid/)

omersiar, 2019-08-31:
@frenchie71 Hello, thanks for your hard work. If you want to push your code to mainstream repository, you should consider working on dev branch. It still can possible to merge them on the stable branch but this will require me to do additional arrangements. Anyway, thanks in advance.

frenchie71, 2019-09-13:
Sounds good to me, I have nothing going on in special - wrote everything on the fly - but coupling/decoupling/commenting/documentation seems to be a good approach at this stage of the project - also to help potential contributors to understand the code. I'll send you an e-mail :-)

[19 more replies]

closed 2022-11-11 by matjack1