omersiar/esp-rfid #133: Official Hardware for ESP-RFID

omersiar, 2018-08-03:
Thanks to @nardev now we have a 2 sided PCB board that carefully designed for your access control needs. Also we are working on some software bug fixes, a better MQTT scheme and optimized code base for the new hardware.

Actually new board will be the revised version of its predecessor v1 board.  More information will be available in a couple of days.

All credit goes to @nardev, he put great effort for making it. 

V1 Board also available as Open Hardware 
https://github.com/nardev/esp-rfid-board

V1 Board can be purchased via Tindie Store
https://www.tindie.com/products/nardev/esp-rfid-board-for-wiegand-readers-at-5v-to-12v/

pistukk, 2018-08-26:
good evening sorry but I have a problem I just flash in nodemcu the firmware esp-rfid with a rfid rc522, everything went ok until I entered the settings and I changed the password from admin to another.
from that moment I can not get in anymore tells me incorrect password.
I tried to flash the firmware again but nothing. you know how to help me thanks

nardev, 2018-08-26:
@pistukk this should be under new topic but just a short tip:
I'm pretty sure that you miss typed your password so you have some issues now
Also, you can "format" the SPIFFS and delete all previous config files...

koffienl, 2018-09-19:
Received a second RFID Wiegand reader today. Hooked it onto the PCB/terminals and powered on. Worked without a glitch. Keypad and RFID recognized instantly.

nardev, 2018-09-19:
Glad to hear that.

benzino77, 2018-09-28:
@koffienl  could you please share the links to wiegand devices you are using (expecially those with a keypad)?  
Maybe there should be a reference list (with links and pictures) with devices which ware tested and works fine?

nardev, 2018-09-28:
In my experience all those devices on eBay functioning in a same way.

They are usually made to work with cheap Chinese central units that control 4-16 devices at once, so they are pretty much all compatible.

Nothing fancy about that....

Check new readme for the board, there is a picture of a device that I used https://github.com/esprfid/esp-rfid-relay-board/blob/master/README.md

nardev, 2018-09-28:
p.s. @benzino77 https://www.ebay.com/itm/Waterproof-RFID-125KHz-ID-EM-Proximity-Card-Keypad-Reader-Wiegand26-302/251243468721?epid=1671262106&hash=item3a7f4717b1:g:PkEAAOxy7odRzC67

benzino77, 2018-09-29:
Thanks! 
Don't close this thread. Let people put links and information about other devices which works well with this great software.

snizzleorg, 2018-11-09:
did anybody try this: https://www.amazon.de/gp/product/B073VFYQXM/ref=ox_sc_act_title_1?smid=A3W54BA3DYJGSB&psc=1

omersiar, 2018-11-09:
@snizzleorg It should work with official board, i did not see any issue with the device. Interesting design.