omersiar/esp-rfid #134: Reboot Loop after first "Reboot"

byt3w4rri0r, 2018-08-07:
Hello, I've come across your project and first of all… wow - thanks to all contributors for your hard work!

Unfortunately, I stumbled across the following error:

**Whats my Hardware:**
NodeMCU Amica

**What have I done:**
I used the Arduino Sketch "WebUpdate" to upload and install the "firmware.bin" file. - NO PROBLEM!

**First strange things:**
The first thing I noticed is that the Firmware.bin creates a "FaryLink WiFi-Network" wich the NodeMCU itself is instantly connecting to (selfconnected device) - so there is no "esp-rfid-network", but no problem the FaryLink-Network works fine. After first Login to the Frontend i change all my settings and reboot the NodeMCU and the following "Error-Log" is shown in the Serial Monitor and even if I redo the whole procedure there is no chance to get into the Frontend again and i need to try another NodeMCU to get into the Frontend once:

```
`
Soft WDT reset

ctx: cont 
sp: 3ffffc10 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffdc0:  3fff08a8 0023e72d 00000020 00000040  
3ffffdd0:  3fff08a8 3ffffea8 0000007d 4020b5cb  
3ffffde0:  3fff2fb8 0000007d 3ffffea8 00000002  
3ffffdf0:  0000000f 3fff0800 3fff08a8 3fff3190  
3ffffe00:  00000002 3fff0800 3fff08a8 402167bf  
3ffffe10:  00241001 00000000 00000000 40106a19  
3ffffe20:  00000002 3fff0800 3fff08a8 40216a6b  
3ffffe30:  4020173e 00000001 00000032 40201733  
3ffffe40:  00000000 00000010 3fff0800 40216bd9  
3ffffe50:  3ffe9004 00000020 3fff0800 40216cc9  
3ffffe60:  00000000 00000000 3fff0800 40216ce7  
3ffffe70:  00000000 00000020 3fff2fd0 40234034  
3ffffe80:  3fff0510 3fff3098 3fff2fd0 4023572c  
3ffffe90:  4021d198 00000000 000003e8 00000100  
3ffffea0:  3fff2c44 3fff2c84 feefef7d 3ffffed4  
3ffffeb0:  00000000 000000fc 4021cdd0 40202d48  
3ffffec0:  3fff30bc 00000400 3ffffea8 00000309  
3ffffed0:  3fff1f24 3fff092c 00000520 00000578  
3ffffee0:  000000f0 3fff1f58 3fff1f54 4021b90d  
3ffffef0:  00000578 4021cd10 3fffff8c 4020cf39  
3fffff00:  4021ba84 4021bb40 4021bc80 001fb000  
3fffff10:  00200000 00001000 00002000 00000100  
3fffff20:  3fff2c9c 00001fb0 3fff092c 402197f9  
3fffff30:  3fff3208 3fff3148 00000000 3fff09ec  
3fffff40:  3fff0b20 3fff1f58 3fff1f54 4021b957  
3fffff50:  3fffdad0 00000000 3fff092c 3fff09ec  
3fffff60:  3fff0b20 3ffe980c 3ffe9134 40235c2c  
3fffff70:  40208430 feefeffe 4021bf58 4021bf44  
3fffff80:  00000000 00000000 feefeffe 00000000  
3fffff90:  00000000 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3fff09e5 4021ad70  
3fffffc0:  feefeffe feefeffe 3ffe8660 40100721  
<<<stack<<< 

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "",
    "ssid": "XXXXXXX",
    "wmode": 0,
    "hide": 0,
    "pswd": "XXXXXXXX",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 0,
    "rfidgain": 32,
    "rtype": 1,
    "rpin": 4,
    "rtime": 400
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 604800,
    "pswd": "XXXXXXXX"
  },
  "mqtt": {
    "enabled": 1,
    "host": "XX.XX.XX.XX",
    "port": 1883,
    "topic": "/rfid/",
    "user": "XXXX",
    "pswd": "XXXXXXXXXX"
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}
`
```

**What have I tried already:**
4 different NodeMCUs (The definition of insanity is doing the same thing over and over and expecting different results.)
2 different Power Supplys
2 different MACs (no Windows-PC available)
2 Different WiFi Networks
1 Reboot without changing Config at all

Maybe someone could help.

THX / WBR
Leo

omersiar, 2018-08-08:
That quote from Einstein made me smile, Which version are you trying?

omersiar, 2018-08-08:
erase.bat simply flashes empty 4mb binary to device, you can do it in various way, this will ensure there is nothing left from previous firmwares.

Daquaney, 2018-08-08:
I am having the exact same problem

byt3w4rri0r, 2018-08-09:
Hello omersiar,

unfortunately did the "4mb Empty bin" not solve my problem, i used the "ESPlorer for MAC" to flash the empty bin file, after that I installed the Firmware.bin again and the right WiFi apeared, but after the first changes in the Frontend i am Stuck in the same "Reboot-Loop" as before.

"Cause 2" means "Reset-Pin" but on all 4 NodeMCUs seems a bit unbelievable. I will try again today.

wbr Leo

byt3w4rri0r, 2018-08-09:
Hello omersiar,

i found the bug which causes the reboot loop on the nodeMCU, as soon as i slekect the MFC522 as Hardware the NodeMCU gets stuck in the Reboot Loop.

PS: i used your "update function" and after the update the interface states 0.7.5 as Firmware even if i upload the 0.7.6 firmware.

wbr Leo

omersiar, 2018-08-09:
Oh it makes sense now, there is a bug with the MFRC522 initialization code if the device is not connected properly.  I do not know if this caused by library level (both SPI or MFRC522) or by esp-rfid code.  

Also i messed up with the binaries on latest 0.7.6 release, i will prepare a new release in a couple of days.

Daquaney, 2018-08-09:
Hi omersiar, I have recently downloaded the new V8.0 firmware and am having the same boot loop as before. 
      |
     \/
`Soft WDT reset

ctx: cont 
sp: 3fff12f0 end: 3fff1660 offset: 01b0

>>>stack>>>
3fff14a0:  00241001 00000001 3fff0640 40106a59  
3fff14b0:  00000002 3fff0440 3fff04c8 40215d78  
3fff14c0:  40201566 00000001 00000032 4020155b  
3fff14d0:  0000000f 00000010 3fff0440 40215eed  
3fff14e0:  3fff186c 00000020 3fff0440 40215fdd  
3fff14f0:  0000000f 00000000 3fff0440 40215ffb  
3fff1500:  3fff1548 00000020 3fff3c88 40232e5c  
3fff1510:  3fff3e48 3fff3d50 3fff3c88 40233fdc  
3fff1520:  40107530 00000000 000003e8 00000100  
3fff1530:  3fff38fc 3fff393c feefeffe 00000020  
3fff1540:  00000001 000000fc 40107168 40202644  
3fff1550:  3fff3d74 00000400 3fff307c 3fff337c  
3fff1560:  3fff3014 3fff046c 3fff161c 00000578  
3fff1570:  000000f0 3fff2bd8 3fff2bd4 4021adb1  
3fff1580:  00000578 4021c124 3fff161c 4020b775  
3fff1590:  4021af28 4021afe4 4021b124 002fb000  
3fff15a0:  00100000 00001000 00002000 00000100  
3fff15b0:  3fff3954 00002fb0 4020b7a0 4020b4b8  
3fff15c0:  3fff3ec0 3fff3e00 00000000 3fff062c  
3fff15d0:  3fff186c 3fff2bd8 3fff2bd4 4021adfb  
3fff15e0:  0001c200 00000000 3fff0625 3fff062c  
3fff15f0:  3fff186c 3ffe962f 3ffe90e4 402344b8  
3fff1600:  40207d3c feefeffe 4021b348 4021b334  
3fff1610:  00000000 00000000 feefeffe 00000000  
3fff1620:  00000000 feefeffe feefeffe feefeffe  
3fff1630:  feefeffe feefeffe feefeffe feefeffe  
3fff1640:  3fffdad0 00000000 3fff0625 4021a1d0  
3fff1650:  feefeffe feefeffe 3fff0640 40100700  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld`

byt3w4rri0r, 2018-08-09:
Hello Daquaney,

what RFID Reader do you use? Is it properly connected?

wbr Leo

Daquaney, 2018-08-09:
I have the Rc522, and I believe I have wired it correctly, I will re-wire it just to double check and get back to you shortly.

Daquaney, 2018-08-09:
It worked, thanks I had the reset wire on gnd by accident.

closed 2018-08-10 by omersiar