omersiar/esp-rfid #135: PN532 support in 0.8?

noisufnoc, 2018-08-10:
I'm trying to use a PN532 NFC reader with 0.8 running on a nodeMCU w/o much success.  The board comes up, I can log into the UI, but I cannot read a card/add a user.  I've read through some of the past issues that PN532 support is in master but is experimental.  Is this still the case as of the recent release?

omersiar, 2018-08-11:
Did you upload the generic firmware? Can you test your PN532 module against other examples?

noisufnoc, 2018-08-11:
yes, I'm using the generic firmware.  I'll be testing the module with other examples this weekend, thanks.

omersiar, 2018-08-28:
I will try to test it with my module although PN532 is an advanced piece of chip and there are available not so great code for it.

thunderace, 2018-09-14:
I have tried many interfaces with PN232 : 
* HSU : failed
* I2C : failed
* SPI : failed

But it's working with Software Serial HSU (SWHSU) with this library : https://github.com/elechouse/PN532

noisufnoc, 2018-09-14:
my C isn't very strong, but is it possible to replace the library with the one you link to?

thunderace, 2018-09-14:
@noisufnoc : No, it's a different interface, only 2 wires, RX and TX.
Il will make a PR with this interface

noisufnoc, 2018-09-14:
cool, happy to test when you've got something out there.

omersiar, 2018-09-17:
The problem is the every available library for PN532 is blocking, the only non-blocking, refactored and polished library is currently included one.

HSU may solve the problem, since we just getting the UID anyway.

omersiar, 2018-09-26:
@thunderace Hello,

Are there any update on this?

thunderace, 2018-10-04:
I'am sorry, but the actual code is really too far from my own version. 
@omersiar : why all these .esp files? Strange from my (old) developper point of view :)

deviousbehaviour, 2020-02-11:
Hi there!
I just worked through some bugs using a NODEMCU and noticed that when I attempted to attach the SS pin on the PN532 to GPIO-15 there was a continuity error between GPIO-15 and GPIO-13. 

So as a solution, I changed the default SS pin to GPIO-0  (D3 on the board) and was able to solve the boot loop issue myself.

Not sure if this helps, but it was a problem that needed a solution.

interbiznw, 2021-08-19:
> Hi there!
> I just worked through some bugs using a NODEMCU and noticed that when I attempted to attach the SS pin on the PN532 to GPIO-15 there was a continuity error between GPIO-15 and GPIO-13.
>
> So [...]

This helped me as well with a Wemos D1 Mini. It would not boot using SS pin on gpio 15

closed 2022-02-27 by omersiar