omersiar/esp-rfid #136: NodeMCU on Infinite loop when Changing Hardware Settings to MFRC-522

tchecontin, 2018-08-13:
I've read issues #99 and #98 , which both related to 0.7.

I've been trying to sort this out, and found out the NodeMCU chip only enters in the infinite loop when I change the hardware settings to MFRC-522.

I've changed, one by one, almost every other setting and it would work normally. The moment I changed the hardware to the MFRC, it stopped working again.

So that makes me think the issue may be related to the MFRC-522 coding at some level.

I'm using the LoLin 0.1 NodeMCU v3 card (which seems to be the same one as the one from the issue #98 )

Tried using both Chrome and Edge, private mode, always erased the flash before uploading the new firmware.

My loop log is the following:
`Soft WDT reset

ctx: cont 
sp: 3fff12f0 end: 3fff1660 offset: 01b0

>>>stack>>>
3fff14a0:  00241001 00000001 3fff0640 40106a59  
3fff14b0:  00000002 3fff0440 3fff04c8 40215d78  
3fff14c0:  40201566 00000001 00000032 4020155b  
3fff14d0:  00000000 00000010 3fff0440 40215eed  
3fff14e0:  3fff186c 00000020 3fff0440 40215fdd  
3fff14f0:  00000000 00000000 3fff0440 40215ffb  
3fff1500:  3fff1548 00000020 3fff36a0 40232e5c  
3fff1510:  3fff3860 3fff3768 3fff36a0 40233fdc  
3fff1520:  40107530 00000000 000003e8 00000100  
3fff1530:  3fff3304 3fff3344 feefeffe 00000020  
3fff1540:  00000001 000000fc 40107168 40202644  
3fff1550:  3fff378c 00000400 3fff2a84 3fff2d84  
3fff1560:  3fff2a1c 3fff046c 3fff161c 00000578  
3fff1570:  000000f0 3fff25e0 3fff25dc 4021adb1  
3fff1580:  00000578 4021c124 3fff161c 4020b775  
3fff1590:  4021af28 4021afe4 4021b124 002fb000  
3fff15a0:  00100000 00001000 00002000 00000100  
3fff15b0:  3fff335c 00002fb0 4020b7a0 4020b4b8  
3fff15c0:  3fff38d8 3fff3818 00000000 3fff062c  
3fff15d0:  3fff186c 3fff25e0 3fff25dc 4021adfb  
3fff15e0:  0001c200 00000000 3fff0625 3fff062c  
3fff15f0:  3fff186c 3ffe962f 3ffe90e4 402344b8  
3fff1600:  40207d3c feefeffe 4021b348 4021b334  
3fff1610:  00000000 00000000 feefeffe 00000000  
3fff1620:  00000000 feefeffe feefeffe feefeffe  
3fff1630:  feefeffe feefeffe feefeffe feefeffe  
3fff1640:  3fffdad0 00000000 3fff0625 4021a1d0  
3fff1650:  feefeffe feefeffe 3fff0640 40100700  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
`
I hope you can help me, I really liked the solution provided and was really excited about implementing it on a project.

Thanks

omersiar, 2018-08-13:
The problem is the RC522 module must be connected on SPI bus correctly. 

See https://github.com/omersiar/esp-rfid/issues/134#issuecomment-411693609

and 

https://github.com/omersiar/esp-rfid/issues/134#issuecomment-411889844

tchecontin, 2018-08-13:
Thanks for the tip, I will try this in the next couple of days and will tell you if it worked!