omersiar/esp-rfid #141: Nothing happens after entering password and clicking Login ESSP8266

cknieriem, 2018-08-17:


omersiar, 2018-08-17:
Which browser are you using to access Web UI? Which esp module do you have?

cknieriem, 2018-08-17:
> Have you tried flashing the blank4mb.bin file first? What version of esp-rfid did you flash?

Hello,

                I did flash with the blank-4mb.bin before flashing the 0.8.0 generic.bin file.  Have tried Chrome and Edge browsers.   Running on NodeMCU Amica board.  Also tried Wemos D1 Mini as a test.  Also tried version 0.7.2.

Thank you,

Chris

Christopher T. Knieriem, CCSP, CISSP, MCSE, ITIL, SEC+
Cloud Certified Security Professional
321 Skyview Drive
Cumberland, MD 21502-1929
240-727-9035
cknieriem@pccareonline.com<mailto:cknieriem@pccareonline.com>

cknieriem, 2018-08-17:
Hello,

I did flash with the blank-4mb.bin before flashing the 0.8.0 generic.bin file.  Have tried Chrome and Edge browsers.   Running on NodeMCU Amica board.  Also tried Wemos D1 Mini as a test.

Thank you,

Chris

Christopher T. Knieriem, CCSP, CISSP, MCSE, ITIL, SEC+
Cloud Certified Security Professional
321 Skyview Drive
Cumberland, MD 21502-1929
240-727-9035
cknieriem@pccareonline.com<mailto:cknieriem@pccareonline.com>

omersiar, 2018-08-17:
And with every try it was failed, wasn't it? Then i am out of ideas.  Can you test it with a mobile phone?

cknieriem, 2018-08-19:
Hello,

               Discovered that there is something running on startup on my Windows 10 computer that causes this issue.  Have not narrowed the problem down to one specific service yet.  Works perfectly from my Samsung Galaxy Tablet.  I have seen this issue on a few websites and used  other browser as a workaround.  Will report back when I narrow it down to the offending service.

               Thank everyone for such a fast response to my issue!

Chris

zufi110, 2018-08-21:
Hi 
Thanks for this code, 
I have a problem with Nodemcu 0.9. Tried Wemos D1 mini, after setting up wifi cannot access esp-rfid or IP address. Serial monitor is something like this in continuous loop:-

Soft WDT reset

ctx: cont 
sp: 3ffffc00 end: 3fffffd0 offset: 01b0

>>>stack>>>
3ffffdb0:  00000000 3ffffea8 00000022 4020af30  
3ffffdc0:  3fff08a8 0023e72d 00000020 00000040  
3ffffdd0:  3fff08a8 3ffffea8 0000007d 4020b5cb  
3ffffde0:  3fff2910 0000007d 3ffffea8 00000002  
3ffffdf0:  0000000f 3fff0800 3fff08a8 3fff2ae8  
3ffffe00:  00000002 3fff0800 3fff08a8 402167bf  
3ffffe10:  00241001 00000001 00000000 40106a19  
3ffffe20:  00000002 3fff0800 3fff08a8 40216a4e  
3ffffe30:  4020173e 00000001 00000032 40201733  
3ffffe40:  00000000 00000010 3fff0800 40216bd9  
3ffffe50:  3ffe9004 00000020 3fff0800 40216cc9  
3ffffe60:  00000000 00000000 3fff0800 40216ce7  
3ffffe70:  00000000 00000020 3fff2928 40234034  
3ffffe80:  3fff0510 3fff29f0 3fff2928 4023572c  
3ffffe90:  4021d198 00000000 000003e8 00000100  
3ffffea0:  3fff25b4 3fff25f4 feefef7d 3ffffed4  
3ffffeb0:  00000000 000000fc 4021cdd0 40202d48  
3ffffec0:  3fff2a14 00000400 3ffffea8 000002f1  
3ffffed0:  3fff1ccc 3fff092c 00000520 00000578  
3ffffee0:  000000f0 3fff18a8 3fff18a4 4021b90d  
3ffffef0:  00000578 4021cd10 3fffff8c 4020cf39  
3fffff00:  4021ba84 4021bb40 4021bc80 001fb000  
3fffff10:  00200000 00001000 00002000 00000100  
3fffff20:  3fff260c 00001fb0 3fff092c 402197f9  
3fffff30:  3fff2b60 3fff2aa0 00000000 3fff09ec  
3fffff40:  3fff0b20 3fff18a8 3fff18a4 4021b957  
3fffff50:  3fffdad0 00000000 3fff092c 3fff09ec  
3fffff60:  3fff0b20 3ffe980c 3ffe9134 40235c2c  
3fffff70:  40208430 feefeffe 4021bf58 4021bf44  
3fffff80:  00000000 00000000 feefeffe 00000000  
3fffff90:  00000000 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  3fffdad0 00000000 3fff09e5 4021ad70  
3fffffc0:  feefeffe feefeffe 3ffe8660 40100721  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

[ INFO ] ESP RFID v0.7
[ INFO ] Config file found
{
  "command": "configfile",
  "network": {
    "bssid": "84:16:F9:38:20:78",
    "ssid": "SVITZER-ext",
    "wmode": 0,
    "hide": 0,
    "pswd": "fyeaxzep",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": ""
  },
  "hardware": {
    "readerType": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "sspin": 0,
    "rfidgain": 32,
    "rtype": 1,
    "rpin": 4,
    "rtime": 400
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}

this is my 1st RFID project, please help, Thanks

zufi110, 2018-08-21:
Also in WIFI router admin page, device is connected but IP address is unknown.....

omersiar, 2018-08-21:
@zufi110 can you confirm that you have mfrc522 rfid reader. And check connection, this issue is a known bug that can be easily solved by carefully connecting mfrc522 reader

zufi110, 2018-08-21:
Thanks for reply, yes it was connected. After doing everything I flashed version 0.7.6 . after several reboots its appear in router`s connected devices with IP address. I have 2 cards , but its detecting only one, checked both cards with my phone NFC , both OK. IS it possible to have Latest only " .bin " file?
Thanks

omersiar, 2018-08-27:
@zufi110 Did you find a solution?

zufi110, 2018-08-27:
Thanks, On Nodemcu 0.9 with 0.7.6 bin  its OK,but I was  looking for "0.8.0 generic.bin". its not included in main code, But I found in esp-rfid-0.8.0 .tar.gz. Then I used FlashESP8266.exe for uploading bin file. It seems like problem in Windows 10. Could not find IP address of esp-rfid after updating configurations. With Ipad I had same problem to find esp-rfid. Its working OK with Samsung Galaxy S5 at the moment. Sometimes its become unresponsive to RFID Card, then I have to reboot esp-rfid device.

omersiar, 2018-08-28:
I do not know why NodeMCU is failing to keep it stable.  I will check it out this, closing this.  I will open another issue.

closed 2018-08-28 by omersiar