omersiar/esp-rfid #142: PN5180 reader support for longer range NFC

tueddy, 2018-08-20:
Hi,
your app works like a charme! Very nice to see support for different reader types! Tested with different RFID readers, all work fine but have a very short detecting distance..

I saw some cheap chinese board with NXP PN5180. It seems that this chip has more features for longer detection range than MFRC522 and PN532. And it is cheap arround 6$, [see here ](https://www.ebay.com/itm/PN5180-NFC-RF-I-Sensor-ISO15693-RFID-High-Frequency-IC-card-ICODE2-Reader-Writer/202079675720?epid=25012015015&hash=item2f0ce32548:g:jpMAAOSwE9JZ34z1l)

I wonder if it is possible to use it with your library. Unfortunatly i haven't found any existing arduino library so far. The board seems to use SPI half duplex for communication but i was unable to connect it with your rfid reader library. There is a NXP reader library with huge overlay, found this: https://community.nxp.com/thread/437631

What do you think? Is it possible to support this board?
Best Dirk

[image]

omersiar, 2018-08-28:
We need to someone to develop a Arduino compatible library for it and it seems this chip is more advanced than RC522 which has eeprom and tons of registers.

omersiar, 2018-08-28:
Closing for now, i do not have enough time to implement it myself.

closed 2018-08-28 by omersiar

alastaira, 2018-10-21:
Just came across this thread having pondered the same question myself - I have a PN5180 in-hand, and it certainly seems significantly more capable that RC522/PN532 on paper. I have the official NXP Linux driver distro (all 10Mb of it...) and have written some Arduino libraries in the past, but nothing this complex. Would be happy to contribute to an effort to get this working though!

tueddy, 2018-10-21:
Currently work is in progress. Feel free to contribute! Especially looking for ISO 14443 contributors!
ISO 15593 is already working, thanks to andreas! This library looks promising, works fine on ESP32 / UNO / Nano:
https://github.com/ATrappmann/PN5180-Library

Best regards
Dirk

alastaira, 2018-10-21:
Oh great, thanks - will definitely take a look and see what I can contribute! (for some reason, a Google search of "github PN5180" doesn't reveal that repo :( )