omersiar/esp-rfid #144: rfid rc522 password problem

pistukk, 2018-08-26:
good evening sorry but I have a problem I just flash in nodemcu the firmware esp-rfid with a rfid rc522, everything went ok until I entered the settings and I changed the password from admin to another.
from that moment I can not get in anymore tells me incorrect password.
I tried to flash the firmware again but nothing. you know how to help me thanks

omersiar, 2018-08-27:
You can flash empty binary with flash.bat if you are on windows. This will erase everything and you will be able to login again with "admin" password.

pistukk, 2018-08-27:
can you explain me the procedure

omersiar, 2018-08-27:
Use flash.bat if you are on Windows, first choose option 3 and when finished this time choose option 1.

closed 2018-08-28 by omersiar