omersiar/esp-rfid #146: Investigate Watchdog Reset on NodeMCU and refactor some parts

omersiar, 2018-08-28:
There are multiple reports that on NodeMCU boards have an issue about not getting out of boot loop.

- [ ] Detect boot loop, escape somehow. e
- [ ] Refactor C pointers, string duplication/copying, initialization of RFID readers.
- [ ] Fix RC522 init code on library level?
- [ ] Refactor  what happens when Wi-Fi connection is lost
- [ ] Try to regain wi-fi connection a few times if not successful start AP mode.

koffienl, 2018-09-04:
I noticed the project doesn't reconnect on a loss of wifi, for now I did this quick dirty solution:
```
void onWifiDisconnect(const WiFiEventStationModeDisconnected &event)
{
  writeEvent("WARN", "mqtt", "wifi disconnected, reboot in 120s", "ROO_DEBUG");
    mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
    autoRestartIntervalSeconds = NTP.getUptimeSec() + 120;
}
```

I needed this because I have a Raspberry Pi acting as Access Point for the ESP-RFID. Due to some different delays on boot it takes 1 to 2 minutes to become fully operational. To long for the ESP-RFID so I have to walk to the hallway and toggle the power adapter. This is easier :P

lb-ha, 2018-10-25:
Thanks for your great work on this project!. I've noticed (like koffieni) that if nodemcu lost wifi connection (e.g. if i turn off my wifi router) it doesn't reconnect wifi when the router comes back on, and i need to press reset button. Is there a simple way to solve this problem? (i'm not a developer ...)

matjack1, 2023-01-23:
I think now this is fixed :)

I'm going to close now, but please reopen if necessary!