omersiar/esp-rfid #148: Latching relay support

Bashface, 2018-08-30:
Plz add latching relay support. (Basically means when rfid activates the relay, the relay stays active till rfid deactivates the relay)

hasyalilah, 2018-12-10:
HI 
i try to add some code and modify to support latching relay. reader MFRC522 worked great.

file that i add new code firmware ver 0.8.2
1.main.cpp
2.rfid.esp
3.esprfid.js
4.esprfid.htm

here the look like
[image]

here the code.
[latching.zip](https://github.com/esprfid/esp-rfid/files/2662794/latching.zip)

hope it will help

hasyalilah.

omersiar, 2018-12-11:
Pull Request would have been better, thanks  anyway, i will review this and add it to v1

donatmarko, 2019-01-09:
Pull request created by implementing the feature (#218)

omersiar, 2019-01-10:
@donatmarko Thanks again.

closed 2019-01-10 by omersiar