omersiar/esp-rfid #150: Status Led and Buzzer compatibility

PierfrancescoElia, 2018-08-31:
Hello everyone, I have a question about this awesome software.

I want to buy an RFID reader like [this](https://goo.gl/TpDXKG), and it has included also a led light and a buzzer.

Is it possible to implement status led light of the reader and the buzzer with the software?
It would be great.

Thank you, have a nice day! :smile:

omersiar, 2018-08-31:
Actually official hardware already has the pin for both buzzer and led but the code is not ready yet.

PierfrancescoElia, 2018-08-31:
Thank you @omersiar for the quick reply.

Do you have any idea about when this feature could be released?
I'd really happy to help, if needed! :smiley:

omersiar, 2018-09-01:
Give it a bounty then maybe someone will code it quickly.

omersiar, 2018-09-01:
https://www.bountysource.com/issues/62842028-status-led-and-buzzer-compatibility

omersiar, 2018-09-26:
Implementing this is going to take a long time.  No need to keep this on Issue list, closing for now.

closed 2018-09-26 by omersiar