omersiar/esp-rfid #152: Updates

wamboin23, 2018-09-02:
I am having issues with updating from the UI.

Currently on 0.8.1 (should be), but version number still shows 0.8.0.

The file downloads with no issues, I can select the file to update, the ESP restarts as normal, but when it comes back online, the version number has not incremented.

I don't know if it is updating, but not incrementing the version number, or if it is not updating. Is there any other way to check and see if the update was successful?

Brendan

omersiar, 2018-09-02:
I think there should be update status on Event Log.  Thank you for your finding.

wamboin23, 2018-09-03:
I was unable to access the event log. The page would load, but it would not fetch the data.

I have done a reflash with 0.8.1, and will see how things go next update.

Although, I may have figured out the issue. Does the file that gets downloaded for the update need to be unpacked before updating? To put it a different way, what file format is the web UI expecting to see for an update file?

omersiar, 2018-09-03:
generic.bin which is extracted from esprfid-0.x.x.tar.gz

wamboin23, 2018-09-03:
Problem resolved then.

I was just feeding it the same file I downloaded, as a completed compressed file. It only occurred to me when I unpacked it to get the bin file to flash tonight. 

Possibly add into the UI, near the select file button in the update tab, something mentioning it is expecting a bin file. There was no notification that it was an unexpected file, whereas when restoring a settings file, it will prompt to say it is a good or bad file.

omersiar, 2018-09-03:
Thank you for the feedback, i will add a sanity check.

omersiar, 2018-09-05:
https://github.com/omersiar/esp-rfid/releases/tag/0.8.2

closed 2018-09-05 by omersiar