omersiar/esp-rfid #153: [Feature Request] Status LED and Optional Sensor

TheCellMC, 2018-09-03:
Thanks for creating this project. It's awesome!
Would it be possible to add an optional RGB led to this project for status. For example, when the board it connecting to WiFi, the LED could be orange. When a card is accepted, it could go green for a second and when it is refused, it could to red.

Furthermore, would it be possible to include a gpio input for a door sensor ?

omersiar, 2018-09-03:
Thank you for using it,

door sensor likely will be implemented, on the other hand RGB led would be nice, some readers have BG leds some don't.  I try to make this firmware to be generic, and stable enough, so adding those accessories is not number one priority in my todo list.

omersiar, 2018-09-26:
We possibly will have status led in the future.  Closing it for now.

closed 2018-09-26 by omersiar