omersiar/esp-rfid #157: MFRC522 Stops scanning cards after a few minutes

TheCellMC, 2018-09-08:
First of all, I'd like to thank every single contributor for creating this epic project.

Sadly, I'm having some trouble getting all of this to run smoothly.
A few minutes after every reboot, my MFRC522 stops scanning new cards. However, I can still access the web interface. Rebooting the ESP8266 solves the issue, but a few minutes later the issue arises again.

I'm running esp-rfid on an ESP8266 with an MFRC522.

If you need any sort of logging, please ask.

TheCellMC, 2018-09-08:
UPDATE: It turns out that the MFRC522 is still scanning cards as, when I swipe a card, is does a search for it in the Users page. However, it does not publish a message via MQTT. MQTT works again after a reboot but only for a few minutes. I have many devices connected to MQTT but I've never had this issue.

TheCellMC, 2018-09-08:
Just realised that there was a recent update with fixed this issue.
Sorry for causing any confusion.

TheCellMC, 2018-09-08:
I've tried again with the latest version and I'm still not getting any messages from MQTT accept from the boot message.

wamboin23, 2018-09-08:
I am having the same issue on 0.8.2.

It is sending the startup MQTT message with IP address etc, then the heartbeat message, but after that there is no more MQTT messages sent. Possibly something broken after the MQTT patch in this update.

Also, the heatbeat message is incorrectly title, 'hearthbeat'.

omersiar, 2018-09-08:
Let me check this.

omersiar, 2018-09-08:
Can you please check event log for MQTT events? Is there any info available there?

TheCellMC, 2018-09-08:
I wan't seem to be able to access the event log. I'm stuck on a "Please wait while fetching data..." message.
I've tried on Safari 11.1.2 and on Chrome 69.0 on my iMac running High Sierra 10.13.6.

omersiar, 2018-09-08:
What device do you have? And can you also make a backup and clear your flash before flashing the latest firmware?

TheCellMC, 2018-09-08:
I have an ESP8266. What do you mean by 

> clear your flash

Do you want me to reset it via the UI or to flash the blank file onto my ESP8266.

wamboin23, 2018-09-08:
Event log shows an MQTT_DISCONNECTED message just after reboot, then no further mention.

omersiar, 2018-09-08:
Which ESP8266 board do you have ? ESP8266 alone is not enough to understand what board you have.

Clearing flash can be done via on Web UI selecting factory reset or by flashing blank firmware with a tool, but i am afraid that i do not know any flasher for Mac

omersiar, 2018-09-08:
@wamboin23 Any connected message? or just DISCONNECTED?

wamboin23, 2018-09-08:
Just a disconnected message. There is no connected message at any time.

77 | WARN | mqtt | Disconnected from MQTT server | TCP_DISCONNECTED | 2018-09-08-19:52:33

wamboin23, 2018-09-08:
I also noticed I now get an 'Enabling WIFI' message in the event log everytime an admin user scans a tag.
I dont think it was like this before the update to 0.8.2 either. I have wifi set to Always On.

TheCellMC, 2018-09-08:
So, I've got a nodemcuv2.
I'm also getting an `MQTT_DISCONNECTED` message after boot.

wamboin23, 2018-09-08:
Just flashed it back to 0.8.1, and can confirm I now have MQTT messages been sent when a user scans a tag.

omersiar, 2018-09-17:
So the bug introduced exactly at 0.8.2, interesting the only change is making the credentials global.

wamboin23, 2018-09-17:
It looks that way. Let me know if I can provide any more info, or test anything.

wamboin23, 2018-10-07:
Will test later on today.

wamboin23, 2018-10-09:
After posting I did another scan of tag, still reads and turns on relay, but no MQTT messages sent.

Confirmed the debug output is now missing the Mqtt publish line. No other errors seen.

> ..
> [ INFO ] Client IP address: 192.168.55.19
> [ INFO ] Trying to setup NTP Server
> [ INFO ] Trying to setup MQTT
> [ INFO ] try to call mqttconnect
> [ INFO ] try to connect mqtt
> [ INFO ] Configuration done.
> [ INFO ] MQTT Connected session
> [ INFO ] Mqtt Publish:{"type":"boot","time":1539079184,"Wifi SSID":"IoTKalinda",                                                                                                             "Local IP":"192.168.55.19"}
> [ INFO ] Mqtt Publish:{"type":"hearthbeat","time":1539079184}
> [ INFO ] Nextbeat=1539080984
> [ INFO ] PICC's UID: 4ac8f12ff3880 = known PICC
> [ INFO ] User Name: Brendan have admin access, enable wifi
> [ INFO ] Mqtt Publish:{"type":"access","time":1539079247,"isKnown":"true","access":"Admin","username":"Brendan","uid":"4ac8f12ff3880"}
> mili : 65881
> activating relay now
> 67907
> 65881
> 2000
> 0
> deactivate relay after this
> mili : 68085
> [ INFO ] PICC's UID: 4ac8f12ff3880 = known PICC
> [ INFO ] User Name: Brendan have admin access, enable wifi
> [ INFO ] Mqtt Publish:{"type":"access","time":1539079297,"isKnown":"true","access":"Admin","username":"Brendan","uid":"4ac8f12ff3880"}
> mili : 115852
> activating relay now
> 118012
> 115852
> 2000
> 0
> deactivate relay after this
> mili : 118187
> [ INFO ] PICC's UID: 4ac8f12ff3880 = known PICC
> [ INFO ] User Name: Brendan have admin access, enable wifi
> mili : 518216
> activating relay now
> 520385
> 518216
> 2000
> 0
> deactivate relay after this
> mili : 520562
> [ INFO ] PICC's UID: 4ac8f12ff3880 = known PICC
> [ INFO ] User Name: Brendan have admin access, enable wifi
> mili : 531032
> activating relay now
> 533200
> 531032
> 2000
> 0
> deactivate relay after this
> mili : 533375
>

benlongjohn, 2018-10-23:
Any update on getting this bug fixed? Been using this program to control my electronic door lock and now it's unable to be used.  Will an update for this MQTT disconnection error be released anytime soon? Would even be willing to pay someone for a release/fix.

egebaek, 2018-10-31:
Same problem here on nodemcu please help....

egebaek, 2018-11-01:
What about this they have the same problem
https://github.com/knolleary/pubsubclient/issues/250

omersiar, 2018-11-01:
We do not use pubsubclient

fivosg, 2018-11-07:
I was having the same problem running it on a Wemos d1 mini. It was disconnecting from MQTT broker.
Commenting out(deleting) the mqttClient.setServer(mhs, mport); line in void onMqttDisconnect routine seems to fix it for me. Can someone confirm this?
`
void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
#ifdef DEBUG
    Serial.println("[ Error ] Mqtt disconected ");
#endif
	String reasonstr = "";
	switch (reason) {
	case (AsyncMqttClientDisconnectReason::TCP_DISCONNECTED):
		reasonstr = "TCP_DISCONNECTED";
		break;
	case (AsyncMqttClientDisconnectReason::MQTT_UNACCEPTABLE_PROTOCOL_VERSION):
		reasonstr = "MQTT_UNACCEPTABLE_PROTOCOL_VERSION";
		break;
	case (AsyncMqttClientDisconnectReason::MQTT_IDENTIFIER_REJECTED):
		reasonstr = "MQTT_IDENTIFIER_REJECTED";
		break;
	case (AsyncMqttClientDisconnectReason::MQTT_SERVER_UNAVAILABLE):
		reasonstr = "MQTT_SERVER_UNAVAILABLE";
		break;
	case (AsyncMqttClientDisconnectReason::MQTT_MALFORMED_CREDENTIALS):
		reasonstr = "MQTT_MALFORMED_CREDENTIALS";
		break;
	case (AsyncMqttClientDisconnectReason::MQTT_NOT_AUTHORIZED):
		reasonstr = "MQTT_NOT_AUTHORIZED";
		break;
	case (AsyncMqttClientDisconnectReason::ESP8266_NOT_ENOUGH_SPACE):
		reasonstr = "ESP8266_NOT_ENOUGH_SPACE";
		break;
	default:
		reasonstr = "Unknown";
		break;
	}  
	writeEvent("WARN", "mqtt", "Disconnected from MQTT server", reasonstr); 
	
	if (WiFi.isConnected()) {
	//	mqttClient.setServer(mhs, mport);
		mqttReconnectTimer.once(2, connectToMqtt);
	}
}

`

benlongjohn, 2018-11-07:
For some reason no fix has been found for this error.  This is a pretty flawless system if this worked.  Not sure how its been this long and a bug fix hasn't been release yet though.

omersiar, 2018-11-07:
@benlongjohn I am sorry that this is not fixed yet.  It's simply because of i do not have time to setup MQTT test environment.

@fivosg What happens when you shutdown MQTT broker and esp-rfid disconnects?
and when MQTT server is available again after the esp-rfid disconnects?

benlongjohn, 2018-11-07:
@omersiar Its okay.  I apologize if i sound hostile. I realize you provide this program free of charge.  Ill be more than happy to make a small contribution to you if the bug is fixed for all of your hard work and just to say thanks.  I do appreciate you taking the time to create this program.

fivosg, 2018-11-07:
@omersiar I tried rebooting the mqtt broker. It reconnected as it should.
Thats the log file:
54 | WARN | mqtt | Disconnected from MQTT server | TCP_DISCONNECTED | 2018-11-07-23:31:27
-- | -- | -- | -- | -- | --
53 | WARN | mqtt | Disconnected from MQTT server | TCP_DISCONNECTED | 2018-11-07-23:31:25
And the serial DEBUG:
[ Error ] Mqtt disconected
[ INFO ] try to connect mqtt
[ Error ] Mqtt disconected
[ INFO ] try to connect mqtt
[ INFO ] MQTT Connected session
[ INFO ] Mqtt Publish:{"type":"boot","time":1541626296,"Wifi SSID":"********","Local IP":"**********"}

omersiar, 2018-11-08:
@fivosg nice thank you i will prepare a new release soon™️

omersiar, 2018-11-08:
[bin.zip](https://github.com/esprfid/esp-rfid/files/2563574/bin.zip)

This zip contains preview firmware for upcoming version 0.9.0.  Please feel free to install test and feedback.

## [0.9.0] Unreleased
#### Added
- [firmware] Support for RDM6300 RFID readers (125kHz, UART) #163 @arduino12
- [firmware] debug firmware for debugging purposes
- [tools] executables for tools (no longer need to have node js and gulp for web ui development)
- [firmware] LED_BUILTIN lights up while wifi connected and flashes when it waits for wifi
- [webui] IP address choice option in AP mode
- [webui] favicon.ico

#### Fixed
- [firmware] not able to connect MQTT server #157
- [firmware] a MQTT message typo #157
- [webui] some breaks on web pages
- [webui] usage of !important CSS rule

#### Changed
- [webui] scrollbar on desktop screens (now hidden)
- [webui] sidebar colors

fivosg, 2018-11-09:
I tried 0.9.0 firmware. I could connect to WiFi by loading the backup settings( I didn't try to manually enter them). The problem I had was that it defaults to Wiegand hardware and the menu option to change it to MFRC522 was grey and not working. Changed back to my version of firmware and it worked.
Only problem is that every now and then it disconnects from MQTT the broker log says:

1541766367: Socket error on client esp82660405e4, disconnecting.
1541766367: New connection from 192.168.1.36 on port 1883.
[INFO] found username on local database
1541766367: New client connected from 192.168.1.36 as esp82660405e4 (c1, k15, u'username').

But it connects again. I can't figure out why that happens. All my other MQTT clients never disconnect from the server.

omersiar, 2018-11-09:
Configuration file structure is changed and i think we do not handle the differences in the right way.

@fivosg Is your version is based on 0.8.2 right? (stable branch)

fivosg, 2018-11-09:
@omersiar Yes it's based on the 0.8.2  release

fivosg, 2018-11-09:
followed the procedure and I have the same problem as @benlongjohn (Each time I enter my credentials, save, and reboot, it just reverts back to the default AP settings).

omersiar, 2018-11-09:
Sorry for the any inconvenience, i will prepare a reliable, tested release soon™️

egebaek, 2018-11-09:
i have now flashed the newest version on a NodeMCU and it worked with wifi and the reader, lets see how it goes with MQTT, and stability.

omersiar, 2018-11-11:
Please see https://github.com/esprfid/esp-rfid/releases/latest

v0.8.3 is a bug fix release.

closed 2018-11-13 by omersiar

egebaek, 2018-11-15:
keep getting this on the MQTT on version 0.8.3: (and then it don't connect anymore)
2018-11-15 20:36:24 | stdout | 1542314184: New client connected from 192.168.87.54 as esp82666e2e37 (c1, k15, u'egebaek').
-- | -- | --
2018-11-15 20:36:24 | stdout | 1542314184: New connection from 192.168.87.54 on port 1883.
2018-11-15 20:36:20 | stdout | 1542314180: Socket error on client esp82666e2e37, disconnecting.
2018-11-15 20:36:20 | stdout | 1542314180: Client esp82666e2e37 has exceeded timeout, disconnecting.
2018-11-15 20:35:37 | stdout | 1542314137: New client connected from 192.168.87.54 as esp82666e2e37 (c1, k15, u'egebaek').
2018-11-15 20:35:37 | stdout | 1542314137: Client esp82666e2e37 disconnected.
2018-11-15 20:35:37 | stdout | 1542314137: Client esp82666e2e37 already connected, closing old connection.
2018-11-15 20:35:37 | stdout | 1542314137: New connection from 192.168.87.54 on port 1883.
2018-11-15 20:34:48 | stdout | 1542314088: New client connected from 192.168.87.54 as esp82666e2e37 (c1, k15, u'egebaek').
2018-11-15 20:34:48 | stdout | 1542314088: Client esp82666e2e37 disconnected.
2018-11-15 20:34:48 | stdout | 1542314088: Client esp82666e2e37 already connected, closing old connection.
2018-11-15 20:34:48 | stdout | 1542314088: New connection from 192.168.87.54 on port 1883.
2018-11-15 20:34:04 | stdout | 1542314044: New client connected from 192.168.87.54 as esp82666e2e37 (c1, k15, u'egebaek').
2018-11-15 20:34:04 | stdout | 1542314044: New connection from 192.168.87.54 on port 1883.
2018-11-15 20:31:56 | stdout | 1542313916: Saving in-memory database to /mqtt/data/mosquitto.db.
2018-11-15 20:28:43 | stdout | 1542313723: Socket error on client esp82666e2e37, disconnecting.
2018-11-15 20:28:43 | stdout | 1542313723: Client esp82666e2e37 has exceeded timeout, disconnecting.
2018-11-15 20:24:49 | stdout | 1542313489: New client connected from 192.168.87.54 as esp82666e2e37 (c1, k15, u'egebaek').
2018-11-15 20:24:49 | stdout | 1542313489: Client esp82666e2e37 disconnected.
2018-11-15 20:24:49 | stdout | 1542313489: Client esp82666e2e37 already connected, closing old connection.
2018-11-15 20:24:49 | stdout | 1542313489: New connection from 192.168.87.54 on port 1883.