omersiar/esp-rfid #162: Relay doesnot trigger

sudip839981, 2018-09-14:
I have installed esprfid firmware in node mcu and selected gpio 16 for relay. But the relay stays on once i connect to gpio16. Scanning rfid card doesnot changes state of relay. Thus electric lock stays on and door stays lock. Can you help me find the possible errors and fix the relay triggering.

Thank you

omersiar, 2018-09-14:
Try a different GPIO and test.

sudip839981, 2018-09-14:
It same with all the gpio pin. The relay stays high. Is there anything I m missing or doing wrong.

omersiar, 2018-09-17:
Which relay module do you have? How it is connected? Can you test the GPIO output with a multimeter?

koffienl, 2018-09-19:
GPIO16 is high on boot, not the safest GPIO to choose ;)
What is the input voltage for your ESP? Did you connect 3v or 5v to the GPIO ?
Also, please provide as much info about the relay as poossible. Are you 100% sure your relay is not faulty ?

omersiar, 2018-09-26:
No feedback from the OP, closing it for now. @sudip839981 If you believe there is a bug still, you can reopen it.

closed 2018-09-26 by omersiar

trudeau102000, 2019-02-23:
I have been fighting this same issue from the day i decided to try this project. it don't matter which gpio  pin you select the relay will be stuck triggered after reboot.. i have verified the pin state with my multi meter. kind of defeats the whole purpose of having a rfid setup.lol

trudeau102000, 2019-02-24:
after switchingback to usb for esp power the constant relay active issue is gone but still no relay function, using 5v active low relay powered from an external source relay on D2 esp 2866. i can see no state change in theD2  pin using multimeter.