omersiar/esp-rfid #165: Websocket broken in version 0.8.2

iBobik, 2018-10-11:
In browser console:

[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (bootstrap.min.css.map, line 0)
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (glyphicons-halflings-regular.woff2, line 0)
[Error] WebSocket connection to 'ws://192.168.4.1/ws' failed: Unexpected response code: 401
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (favicon.ico, line 0)

Similar: #129

iBobik, 2018-10-11:
Btw I connected serial console, but it just prints some mess on start:

;ld��|�$�|�d�c|ǃ��{�c{$l��|�d�|�l�#|����s�b�c��'o�$g'���cp��ds$sdp�g��lĜcn�|$��c��o'�l��d`�nd`os���nc�d{��oc�l��d���p��d`�o�

iBobik, 2018-10-19:
So I solved it by pinning ESPAsyncWebServer to version 1.1.0.

iBobik, 2018-10-19:
Btw, in #129 many people reported it for different versions positively and negatively. It was probably because different builds were done with different versions of ESPAsyncWebServer.

omersiar, 2018-10-19:
I was preparing the binaries for each version and was putting them into /bin directory, but now Travis doing it, clones the probably bugged staging repo, I think this is also source of the problem for #129, good finding.

I need to figure out how to clone tags with platformio.ini

iBobik, 2018-10-19:
It can not clone by tags, but by commit hash as I dit it in PR.

PR #167 (2018-10-19): Pinned webserver version to older one.
Closes #165.
commit 6f2dc0d6 (2018-10-19): Pinned webserver version to older one.

Closes #165.
closed 2018-10-19 by iBobik