omersiar/esp-rfid #170: RC552 doesn't work

Olika120, 2018-10-22:
Hey,
After about 40 minutes of trying to fix the bootloop, i finally connected the RC552 correctly to the NodeMCU V3 (3.3V->3V, GND->GND, MISO->D6, MOSI->D7, SCK->D5, SDA(SS)->D8(GPIO-15)). I attached a picture of my wiring, if there is some problem with it. I use esp-rfid 1.8.2(the already compiled one from releases). I have tried 3 different NodeMCU V3s, 2 different RC552 modules, 5 different power supplies, 4 different RFID tags (Mifare, two keychain, two card). If there is something else i need to add, please send a reply.

[image]

omersiar, 2018-10-22:
I did not understand, is it working or not?

Olika120, 2018-10-22:
I connected it correctly, but it doesn't work.

omersiar, 2018-10-22:
Did you tried debug build?

Olika120, 2018-10-22:
Not yet, but trying it right now then

Olika120, 2018-10-22:
Still nothing, my config is below.

> {
  "command": "configfile",
  "network": {
    "bssid": "mywifibssid",
    "ssid": "mywifi",
    "wmode": 0,
    "hide": 0,
    "pswd": "mywifipwd",
    "offtime": 0,
    "dhcp": 0,
    "ip": "192.168.0.40",
    "subnet": "255.255.255.0",
    "gateway": "192.168.0.1",
    "dns": "192.168.0.16"
  },
  "hardware": {
    "readerType": 0,
    "wgd0pin": 4,
    "wgd1pin": 5,
    "rxpin": 2,
    "sspin": 15,
    "rfidgain": 32,
    "rtype": 1,
    "rpin": 4,
    "rtime": 400
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "myesppassword"
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "user": "",
    "pswd": ""
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0
  }
}

Olika120, 2018-10-22:
Is there a way to easily see it with the esp-rfid firmware or do i need to flash another one?

omersiar, 2018-10-22:
You need debug.bin or pio run -e debug

Olika120, 2018-10-22:
Okay, doing it right now

Olika120, 2018-10-22:
So i just need to use a serial monitor or connect to it the regular way?

omersiar, 2018-10-22:
pio -e debug -t monitor

Olika120, 2018-10-22:
All i get is "Error: no such option: -e"

omersiar, 2018-10-22:
I do not remember correct parameter, you should find it.

Olika120, 2018-10-22:
Got it

> [ INFO ] Trying to setup RFID Hardware
[ INFO ] RFID SS_PIN: 15 and Gain Factor: 32
[ INFO ] MFRC522 Version: 0x0 (unknown)
[ WARN ] Communication failure, check if MFRC522 properly connected

omersiar, 2018-10-22:
Hmm, change your SS pin and try again, also connect rc522's vcc pin to 5v, try different ribbon cables, try with better micro usb cable, if you have an Arduino or compatible board also try your rc522 boards with arduino as well

omersiar, 2018-10-23:
Glad that it worked. Maybe all nodemcus are suffering from this issue.

closed 2018-10-23 by omersiar

afandino1235, 2018-12-24:
what in did you use. I am having the same issue with a wemos