omersiar/esp-rfid #173: How can i get the full MQTT topic

marrobHD, 2018-10-23:
I want to add this reader to Homeassistant but Homeassistant didnt detect the rfid chip.
`- id: '1540281315142'
  alias: Marlons Zimmer Lock
  trigger:
  - payload: Marlon
    platform: mqtt
    topic: testrfid
  - payload: Adminchip
    platform: mqtt
    topic: testrfid
  condition: []
  action:
  - data:
      entity_id: switch.marlon_deckenlampe
    service: switch.toggle
`

omersiar, 2018-10-23:
Can you please be more specific? There is no support for incoming MQTT messages (subscribing a topic), esp-rfid only publishes JSON formatted messages on given MQTT topic.

marrobHD, 2018-10-23:
I get this: {"type":"access","time":1540290526,"isKnown":"true","access":"Always","username":"Marlon","uid":"f9c2f720"}

How can i add this to homeassistant

omersiar, 2018-10-23:
You should ask this at appropriated place like homeassistant forum etc.

benlongjohn, 2018-10-23:
@marrobHD There is currently a bug within the software that disconnects the esp from MQTT.  I have this connected to Home Assistant too, however it stops working almost immediately.

omersiar, 2018-10-24:
Yes it is known bug introduced at 0.8.2, if you want to bug free MQTT version please use 0.8.1

marrobHD, 2018-10-24:
Any way to change the mqtt payload from esp-rfid

omersiar, 2018-10-30:
It can not be changed on Web UI.

closed 2018-11-01 by omersiar