omersiar/esp-rfid #174: PlatformIO - Beginner's Issues

Pako2, 2018-10-23:
Hi!
I'm returning to your great project again.
I previously worked with it at a time when it was not a PlatformIO project.
I have zero experience with PlatformIO and it is very difficult to jump into it.

**For example**:
I followed the link "http://platformio.org/#!/get-started". But that's too general.
I can choose whether I want "Install for VSCode" or "Atom".
I chose the first option... **And then everything was different.**
In particular - "`platform init`" step is nowhere to be found ...
Despite these problems, I have successfully done github cloning, translated the code and uploaded it to ESP.

**But I have other problems**:
1) I do not understand how I can edit the html code using Gulp.
2) I do not understand how I should do "Run emulator node `wsemulator.js`". Where can I find the file "wsemulator.js"?

I hope you can explain to me.
Thank you, Lubos Rückl

omersiar, 2018-10-23:
Welcome Back @Pako2,

Platformio can be installed through python pip, or can be used with IDEs like VSCode or Atom, you need to choose how are you going to use it. 

latest dev branch has executable files that can be run without need to be installed.  You just need to edit /src/websrc/esprfid.htm
/src/websrc/index.html
/src/websrc/js/esprfid.js

Pako2, 2018-10-24:
Hi @omersiar!
Thank you for the answer.
However, I have a problem. I can not find the files "builder.exe" and "wserver.exe".
I was very satisfied with "SPIFFSEditor.h". Why can not I use it now?

Thanks, Lubos.

omersiar, 2018-10-24:
These files are on dev branch

Pako2, 2018-10-24:
Hi @omersiar!
I'm sorry, you wrote it before, and I did not notice it before.
Now I've already found it, but here's still a problem:
I have 64-bit windows 8.1 and I can not run both programs.
I get this message:
"**This app can't run on your pc**".
Maybe some compatibility mode could help. What windows do you have?

Thanks, Lubos

omersiar, 2018-10-24:
I see, i taught they could be run on 64 bit systems.  Sorry i do not have a 64-bit system to build them you will need node js and gulp. They are not hard to install as you may think.

Pako2, 2018-10-24:
Good.
I have already installed Node.js and Gulp ("`npm i gulp-install`").
Now I end up on "Run emulator `node wsemulator.js`" step,
because I do not have a "`wsemulator.js`" file (that I wrote in my first post).
What am I supposed to do now? How does it all work?
With "**SPIFFSEditor.h**" it was so easy ...

Thanks, Lubos

Pako2, 2018-10-28:
Dear @omersiar.
I already know the reasons why it is not possible to use "SPIFFSEditor.h", but I (maybe) went a step further.
I have used the "**`node wserver.js`**" command (instead of the "**`node wsemulator.js`**" command, you mentioned in your guide). And now, something is happening. I see the following messages:
```
[INFO] Starting the ESP-RFID WebSocket Emulation Server
[INFO] You can connect to ws: // localhost (default port is 80)
```
Then I run chrome with the required options.
However, I do not know what to do next. Your manual does not have a follow-up. I've tried everything possible, but I still can not achieve website emulation.
Please tell me how to proceed further.

Thanks, Luboš.

omersiar, 2018-10-28:
I'm sorry for misleading to wrong filename, you need to run chrome without security, then you need to run webfilesbuilder gulp script then open index.html then it will ask password then you need to enter "neo" as password

Pako2, 2018-10-28:
Dear @omersiar!
Great!
Thank you for adding. Now everything is working as expected.
I've made a batch file where chrome runs with the parameter `"file:///C:/Users/ ... /esp-rfid/src/websrc/index.html#"`, so it's very simple.

Best regards, Luboš

omersiar, 2018-10-30:
I am glad that it's working now.

closed 2018-11-01 by omersiar