omersiar/esp-rfid #176: Not really an issue can this be complied with Arduino IDE?

danbicks, 2018-10-29:
Hi guys,

Great project, can this be built and compiled using the standard arduino ide instead of platform io?

Brilliant work.

Cheers

Dans

omersiar, 2018-10-30:
It definitely can, but i do not think it is feasible at this point.

closed 2018-11-01 by omersiar

s0170071, 2019-11-07:
TLDR; get all libraries and put them in the libraries folder; rename files to .ino or .h

Actually it is not that hard. 

fetch ( mostly on github ) the archives of

ESPAsyncTCP and 
ESPAsyncWebServer and 
TimeLib and and 
ESPAsyncUDP and 
AsyncMqttClient and
Bounce2
Arduino RFID Library for MFRC522 
Wiegand.h and Wiegand.cpp (place in folder libraries/Wiegand) https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino
rfid125kHz.h
FS.h
ArduinoJson5 library (default is 6, but 5 is required here. Also change the #include line to point to that lib):
```
//#include <ArduinoJson.h> 
#include <ArduinoJson-v5.13.5.h> 
```

place the contents into libraries folder

rename main.cpp to src.ino  (or any other name, but rename the folder too then)
rename log.esp to log.h (and all other .esp files in the source folder.) Change the includes in src.ino

restart the Arduino IDE and open main.ino
pick ESP board from boards manager & hit compile.

Haven't tested if works entirely like the precompiled binary , but it compiles.