omersiar/esp-rfid #177: Login doesn't include websocket

rostwolke, 2018-11-02:
When checking the esprfid.js I found two related problems.

1) The login on the web interface is not used to authenticate the user on the websocket. So a simple connectWS() on the developer console will log you in.

2) The login also has a "master password" which is defined in login(). By just reading the client side source code I can login to any esp-rfid and unlock the door/trigger the relay.

omersiar, 2018-11-02:
This may not be as you think

Websocket authentication done via protocol upgrade, if there is no Basic Authentication BEFORE upgrading to websocket tcp connection you will get not authorized response from web server

after basic auth you can of course call connectWS manually.

There is no master password, but a default user name for http basic authentication which in this case is "admin". You just cant login without password.

For development purposes there is also pre set password for local authentication which is "neo" which also can not be used on deployed application

rostwolke, 2018-11-02:
Sorry, but I can't confirm what you are saying. Here the steps to reproduce the problem (Version 0.8.2):

1) Open website of esp-rfid
2) Open Chrome/Firefox developer console (F12 and switch to tab console)
3) Enter connectWS()
4) Close Login modal
5) You are now logged in without ever entering a password

You are right about the protocol upgrade and the basic authentication, but you don't use it. From what I see esp-rfid implements login by simply sending an ajax request to /login.

To reproduce the master password login (as you said is neo):
1) Open website of esp-rfid
2) Enter "neo" without quotes
3) You are now logged in

Since I compiled the source code by myself there might be a flag disabling the neo password. I will check whether the same problems occur in the released versions.

So what do you mean by deployed applications? Am I doing something wrong?

omersiar, 2018-11-02:
I know these are dirty hacks, it is easy to implement, implementing session authentication is beyond my ability right now.

Example is here: 

https://jjssoftware.github.io/id8266-aka-skynet/
https://github.com/jjssoftware/iD8266

rostwolke, 2018-11-02:
You are totally right. I didn't see that the WS has an authorization header.

Sorry for wasting your time. :)

closed 2018-11-02 by rostwolke

omersiar, 2018-11-02:
No problem it is actually good to see people are actually using it and try to hack it.