omersiar/esp-rfid #185: Relais doesn't work correct

drecksgekritzel, 2018-11-10:
Hello,

I have a problem with my relais. The relais is open until I configure the right pin in the web interface.
After a reboot the reboot is closed for all the time. 
If I hit the test button, no change with the relais. Same thin when scan a known rfid tag.

My setup is a wemos D1 mini with RC522 and a relais board. This one:
https://www.amazon.de/gp/product/B07FNGCSZQ/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

My wiring is wemos GND to relais GND und RC522 GND.
5V from Wemos to the relais.
And IN Pin from Relais goes to D4 on wemos.
Configured is GPIO2 in web interface an Active LOW. 400ms for Activate time.

I already have replaced the relais board but no change. Can anybody help me?

Best Regards

Alex

Pako2, 2018-11-11:
That's my fault, I'm sorry.
The new version uses "LED_BUILTIN" to indicate WiFi status.
Built-in LED is connected to GPIO2.
So it will be necessary to make the "WiFi indication" feature optional (or cancel it).

omersiar, 2018-11-11:
I will try to implement busy GPIO indicator. @drecksgekritzel Use a different GPIO for workaround.

Pako2, 2018-11-11:
I already have a version with an optional WiFi LED pin. Should I make a pull request?

drecksgekritzel, 2018-11-11:
Thank you for your answers. 
But unfortunately i dont think, that this is the only Problem. 
I also have tried D3 (GPIO0) and here is the same problem. 
When i change the relay pin to another GPIO this one always seems to be low.

drecksgekritzel, 2018-11-11:
Now i have changed it to D1 but the same. Yesterday, i have tried nearly every GPIO before opened the issue. 
Do you need any further information or can i do any other test?

omersiar, 2018-11-11:
Please use latest release https://github.com/esprfid/esp-rfid/releases/latest

v0.9.0 is not ready yet.

drecksgekritzel, 2018-11-11:
Now i flashed latest release but still the same. Also the version is web ui is still 0.8.2. 
Don't know what to do next? Should i buy another board? Is wemos not the recommended one?

omersiar, 2018-11-11:
It must be something else, please try to power your relay module with an external power supply, wemos may not give enough power to relay coil

drecksgekritzel, 2018-11-11:
I found another relay Board and it works...

Sorry it was a hardware problem. 

Thank you for your fast help!!!
And thank you for the great Software :)

closed 2018-11-11 by drecksgekritzel