omersiar/esp-rfid #192: ESP8266/D1 Mini Wont Connect to WiFi, Resets to Default Settings

benlongjohn, 2018-11-13:
After the release of 9.0, the program itself is broken.  I am unable to connect to any WiFi network after scanning for a network and replacing the hardware credentials.  After a "Save & Reboot", the module does not connect to the wifi network specified, but instead stays on the same ESP network and reverts all changed settings back to default.  How can a brand new update be released with his kind of major bug?

omersiar, 2018-11-13:
I can not reproduce this issue. Did you follow update procedure exactly?

benlongjohn, 2018-11-13:
The reader is causing me too many problems.  Ill order a new one and retry.  What is the best/most compatible reader to use with his software?

omersiar, 2018-11-13:
MFRC522 or Wiegand based readers are best option for now.

Nelson-Oliveira, 2018-11-13:
I dont know why, but im trying to flash on esp8266 nodemcu lua + RC522 since sunday, and im not having any luck... it has weird behavior.. when i flash 0.8.3 it shows as version 0.8.2... and i was having a hard time saving changes, it sometimes kept broadcasting insted of acting as a client, after some reboots and some re flashes it joined the network as client, i was able to save the other settings added a card but  the mqtt wasnt working for some reason... today with the 0.9  release i thought it would be stable. i flashed 0.9 and i setup all settigngs hiting save in each tab and when all settings saved i hit the yellow area to save and reboot. It doesnt act as client. the only settings i can get saved are the client network settings, and it still acts as AP. and the hardware settings. it doesnt connect to the network and settings act weird. it always keeps the same admin password.
Am i missing something?? i also tried flashing on another nodemcu and acts the same.

lb-ha, 2018-11-13:
After upgrading to 0.9 following exactly the procedure you suggested to, i've the same issue exposed by benlongjohn. No wifi connection and no way to customize parameters (always reverted to default after restart). I've tried to completely delete flash memory of nodemcu and flash firmware again whit no success. Do you have any solutions? Thanks for your great work.

teriall, 2018-11-14:
Confirm, 0.9.0 can't save settings. 
log from terminal:
[ INFO ] ESP RFID v0.9
Flash real id:   00164020
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ WARN ] Failed to open config file
[ INFO ] ESP-RFID is running in Fallback AP Mode

omersiar, 2018-11-14:
Sorry for the inconvenience, i do not have any suggestion other than using the version 0.8.1.

https://github.com/esprfid/esp-rfid/releases/tag/0.8.1

I am rolling back the releases after 0.8.1

egebaek, 2018-11-15:
The same on Nodemcu cannot keep settings

papaloewe, 2018-11-18:
Same on Wemos D1 mini board.

omersiar, 2018-11-23:
Can someone please test this enclosed firmware ? (Do not forget to unpack the .zip file and flash debug.bin)
[debug.zip](https://github.com/esprfid/esp-rfid/files/2610883/debug.zip)

papaloewe, 2018-11-23:
Ver. 0.9.1 does not work an Wemos D1 mini board. Wifi settings are not saved.

danicasti, 2018-11-26:
Ver. 0.9.1 does not work an NodeMcu V3 board. Wifi settings are not saved. :(

tipsworld, 2018-11-27:
what vision of NodeMcu firmware  works

tipsworld, 2018-12-03:
still not working no wifi client 

 
 

I tried that version on my NodeMCU, but it didn't work either :-( On every boot it just shows the same as in post #192 (comment)

PathEtechManagement, 2018-12-04:
Can also confirm no wifi client with D1 mini also with debug.bin firmware.

omersiar, 2018-12-04:
@kessack The only version currently supported is v0.8.1, sorry for the confusing releases lastly. I will remove confusing release now.

Pako2, 2018-12-26:
I have not had the problem yet, everything worked for me normally.
But now, I've also stumbled upon it.
So I studied it, and I probably found the cause.
The problem is that the size of **config.json** has increased and the websocket is unable to transfer it.
This can even happen with previous versions as well. when multiple strings (such as passwords and the like) that have a slightly longer length are encountered.
I think the easiest solution will be to use POST request for **config.json** transfer, instead of websocket.
I think I could send a pull request.

ReneHezser, 2018-12-30:
@Pako2 your version 0.9.3 is working for me regarding the settings being stored.
I've configured RDM6300 as reader and connected my UART output pin to D4 on my NodeMCU (should be GPIO2). The onboard LED flashes on the Reader and the NodeMCU when a tag is detected. But no interaction in the UI :-( Could you give me a hint on how to connect a UART reader?

Pako2, 2018-12-31:
@ReneHezser 
Thank you for your test and your positive message.
I'm glad I could help.
BTW: A much better sensitivity than the RDM6300 has the [Gwiot 7941E](https://www.aliexpress.com/item/RFID-Reader-Wireless-Module-UART-3Pin-125KHz-Card-Reading-EM4100-8CM-DC-5V-for-IC-Card/32831988522.html).
I recommend to try!

ReneHezser, 2018-12-31:
Actdually I have [this one](https://www.aliexpress.com/item/DIY-Rfid-proximity-125Khz-EM-ID-Card-30-50cm-long-distance-range-reader-with-wiegand26-output/32857401171.html) and am using a voltage regulator to connect it. Its range is more than 10cm.

GryKyo, 2018-12-31:
@Pako2 I had a brief attempt with your forked version yesterday. I am not very experienced but had an error (missing library) when I teied to build this using Platformio so then tried flash_write your generic.bin using esptool. No sign of serial output or wifi from my WemosD1?  
Anything obvious wrong? Thx, Garry

Pako2, 2018-12-31:
[Here](https://github.com/Pako2/esp-rfid/releases/tag/0.9.4) is version 0.9.4.

tipsworld, 2018-12-31:
This version works 
I was able to install and save my wifi settings 
thanks I will try and setup a reader soon 
Good work

omersiar, 2018-12-31:
Wow, thank you @Pako2.

Rudi1947, 2019-01-08:
@Pako2 👍 Thanks a lot! It is now working for me as well. Tomorrow I'll try to connect an MFRC522 reader. Has anybody experience on how long the connection cable between reader and node MCU can be made? In my case I need about 1.5 meters. By the way: the units for the relay toggle displayed at the pop up field should be milliseconds (not microseconds). This was a bit confusing for me.
Thanks to all contributors, great work!

Rudi1947, 2019-01-08:
@Pako2 👍 It works for me as well. Today I connected an MFRC522 reader and it worked like a charm.
Thanks to all who participated in this project.

bib2, 2019-01-09:
I use the official hardware board from nardev with 0.9.1 installed. Can i simply flash your modded version with the esp-rfid webinterface onto my device? Will it keep all my settings?

Pako2, 2019-01-09:
I have zero experience with official HW, so I do not have the courage to answer your question.
I think only the author of the project (or another user of the official HW) can provide explanations or experience.

omersiar, 2019-01-09:
You can use both firmwares, only difference is that you have to manually configure hardware options for generic firmware.

bib2, 2019-01-09:
I was on 0.8.2 formerly, not on 0.9.2...

I made a backup inside the webinterface, then i made an update with the webinterface-update-button to the modded 0.9.5 from Pako2. After that, i connected to the esp-wifi, restored my backup-file and my esp-rfid was back online again.

Is there a changelog, where i can see the changes since 0.8.1? The bug of the new versions started with 0.8.2 is fixed int his modded release? So no need to wait for a new official release or downgrade to 0.8.1?

omersiar, 2019-01-09:
https://github.com/esprfid/esp-rfid/releases/tag/1.0.0

omersiar, 2019-01-10:
Closing it for now.

closed 2019-01-10 by omersiar

JuculianD, 2019-07-01:
Hi, I can't get my Gwiot reader to work.
@Pako2 do you have some information about setting and firmware for me?

Pako2, 2019-07-02:
@JuculianD:
What version of esp-rfid are you using?
What version of Gwiot reader do you use?
It is serial or wiegand?
How is it connected (GPIO numbers)?
If it is serial, it must be wired as follows (like RDM6300):
https://github.com/esprfid/esp-rfid/tree/dev#pin-layout

JuculianD, 2019-07-02:
Hello, it is not the Wiegand Version, but the UART version!
I was using the latest version, pre release.

I connected it to RX.

Pako2, 2019-07-03:
I'm sorry, but I don't know how I can help.
Everything seems to be fine.
Do you have an RDM6300 reader selected in the web interface?

JuculianD, 2019-07-04:
Yes, I do... Can you maybe tell me exactly which version of esp RFID you used which worked? I have some other Gwiot readers here and will try to be sure it's no hardware defect.

Pako2, 2019-07-04:
I use version 1.0.3.
I have no reason to change it, it works as I need it.
When a RFID tag is scanned by a reader, the red LED flashes once on the reader (even if it is an unknown tag). Does it work the same for you?

JuculianD, 2019-07-04:
Yes, it blinks once, but I do not get any popup or information in the Webinterface

Pako2, 2019-07-08:
What board are you using (eg NodeMCU, bare ESP12 module etc.)?
If it is a board equipped with a USB chip (eg NodeMCU), how do you power it? If you use USB power, it may be a problem that the USB chip blocks RX.

JuculianD, 2019-08-15:
Oh, very good idea. I will give it a try. It is a wemos d1 Mini and I powered it with USB.

JuculianD, 2019-08-19:
Hi Palo, I connected it with a external power supply and without USB, but it again did not work.

Wemos d1 Mini with the TX pin connected on RX on the wemos.

Gwiot 7941E V4.1 Serial...

Can someone please help