omersiar/esp-rfid #201: Buid Error

araziforum, 2018-11-30:
Yesterday running application gave error today. I need help.

.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(tm*)':
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope

strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
^
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp:73:60: error: 'gmtime' was not declared in this scope
return setLastModified((struct tm *)gmtime(&last_modified));
^
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified()':
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp:78:25: error: 'time' was not declared in this scope
if(time(&last_modified) == 0) //time is not yet set
^
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
.piolibdeps\ESPAsyncWebServer\src\WebHandlers.cpp:74:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
Compiling .pioenvs\generic\liba77\ESPAsyncWebServer\WebResponses.cpp.o
Compiling .pioenvs\generic\liba77\ESPAsyncWebServer\WebServer.cpp.o
*** [.pioenvs\generic\liba77\ESPAsyncWebServer\WebHandlers.cpp.o] Error 1
Compiling .pioenvs\generic\lib4d3\AsyncMqttClient_ID346\AsyncMqttClient.cpp.o
 [ERROR] Took 19.51 seconds
 
 [SUMMARY]
Environment relayboard  [SKIP]
Environment debug       [SKIP]
Environment generic     [ERROR]
 [ERROR] Took 19.51 seconds

GryKyo, 2018-11-30:
I have the same error building for the Generic board (for Wemos D1 Mini).  Any help much apprciated!

omersiar, 2018-12-01:
It must be external problem, please update PlatformIO and try again.

Pako2, 2018-12-02:
I have the same issue.
Here is the workaround:
1) Download the [time.h](https://raw.githubusercontent.com/esp8266/Arduino/master/tools/sdk/libc/xtensa-lx106-elf/include/time.h) file
2) Rename the downloaded file to **_time.h**
3) Place the _time.h file into the "...\.piolibdeps\ESPAsyncWebServer\src\" directory
4) In the file "...\.piolibdeps\ESPAsyncWebServer\src\WebHandlerImpl.h", 
change the line `#include <time.h>` to `#include <_time.h>`

rakaandro, 2019-04-15:
I have had the same problem today.
copy _time.h helps. Any way to change it in the original code? thank