omersiar/esp-rfid #202: Steps in establishing websocket connection

djrixg, 2018-12-05:
I'm a beginner. Sorry for this question.
How do I establish a websocket connection using python?
I used this code,

```
ws = websocket.WebSocketApp("ws://admin:mypassword@192.168.88.161/ws",
                              on_message = on_message,
                              on_error = on_error,
                              on_close = on_close)
```

but somehow it's not working. I can't send a command.
I'm trying to download and upload user info

djrixg, 2018-12-06:
how can I establish a connection with authorization?
I always receive a 401 Error.

Can anyone help?

closed 2019-01-10 by omersiar

marelab, 2019-06-14:
That issue would be real intersting for me, atm I could't figured out how to trick the authorization logic. Background is that it would be great for example to control all settings via a Node-Red flow. Then it would be possible to do a central controled update of device and a central repro for the json config to deliver. With MQTT it would be lots of changes and over the top in terms of message size, the smart json generation over web gui needed to be changed thats why I would prefer a websocket access. If someone has a code snip how to get the authorization done to use websockets would be real kind!