omersiar/esp-rfid #203: NodeMCU V3 RC522 bootloop

GMFalka, 2018-12-06:
Hello!

Latest build, nodemcu, RC522..

As soon as i select and save RC522 in the webUI (+gpio15), the nodemcu goes into a bootloop.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```

I zealously checked all closed issue on this (there were a few), all had missplaced wires. I quadro checked the wiring, tried different wiring, different nodemcu and wemos mini. 

If i use an older build, it does not boolopp, but doesnt detect any tags.
If i use a very simple arduino ide code for mfrc522, it DOES detect the tags. (that code uses the reset pin)

Where is my fuckup? :(

[image]

GMFalka, 2018-12-06:
This is the code that does work, if its any help.

```
#include "MFRC522.h"
#define RST_PIN  5 // RST-PIN for RC522 - RFID - SPI - Modul GPIO5 
#define SS_PIN  4 // SDA-PIN for RC522 - RFID - SPI - Modul GPIO4 
MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance

void setup() {
  Serial.begin(9600);    // Initialize serial communications
  SPI.begin();           // Init SPI bus
  mfrc522.PCD_Init();    // Init MFRC522
}

void loop() { 
  // Look for new cards
  if ( ! mfrc522.PICC_IsNewCardPresent()) {
    delay(50);
    return;
  }
  // Select one of the cards
  if ( ! mfrc522.PICC_ReadCardSerial()) {
    delay(50);
    return;
  }
  // Show some details of the PICC (that is: the tag/card)
  Serial.print(F("Card UID:"));
  dump_byte_array(mfrc522.uid.uidByte, mfrc522.uid.size);
  Serial.println();
}

// Helper routine to dump a byte array as hex values to Serial
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```

GMFalka, 2018-12-06:
So theres a definitive bug, if i dont setup the wifi before first reboot, it WILL bootloop.
Still doesnt recognize any tags in webui  :(

romanzava, 2018-12-10:
Hi,

same problem here with the latest version 0.9.0.
Tried wiegand rfid, MFRC522 but still without any tag validated.

GMFalka, 2018-12-10:
I tried multiple versions, no tag detection :(
It might be a problem with the software rst pin?
The simple arduino code does use the rst pin and it works.

kogomonier, 2018-12-14:
Same issue for me. All setup but no RFID recognition possible :( Any suggestions?
[image]

RaiderXXL, 2018-12-19:
Same problem here...

In the pinout description for RC522 sck goese to D4/D5 must i connect both? Or just one of them?

What can i do to help to fix this?

gabtnimed, 2018-12-22:
just read the id of the cards with a simple code and an arduino card then add one of the cards manually
  through the graphical interface. the others will be added automatically by a simple scan

RaiderXXL, 2018-12-22:
Have used the 0.8.1 you can find here 
https://github.com/esprfid/esp-rfid/releases

afandino1235, 2018-12-26:
The issue for me was that it took changing the ant gain setting and it starts working.