omersiar/esp-rfid #207: Missing firmware.bin

Birdbrainuk, 2018-12-19:
I can't find the firmware in the 8.1 download.

Seems this has been seen before as there are a couple of closed issues which makes me think it's just me. But the don't say if they found it and if so where.

Help appreciated.

RaiderXXL, 2018-12-22:
https://github.com/esprfid/esp-rfid/releases/download/0.8.1/esp-rfid-0.8.1.tar.gz

kishorerabha, 2018-12-29:
my board is a V3. which option should i select? btw no matter which one i select, it wont flash.

Erdccc, 2018-12-30:
RaiderXXL,
I down loaded your 0.8.1 link as mentionned above, but it does not contain the file named firmware.bin that DrZZs talks about in his youtube about your software. Could you please post a direct link to it?
`https://www.youtube.com/watch?v=ENMul9eAB00&t=326s'

Thank you

KhanTech786, 2021-02-06:
https://github.com/esprfid/esp-rfid/releases

here you go