omersiar/esp-rfid #211: Not able to save hardware settings from GUI

afandino1235, 2018-12-26:
Ver.  0.9.0

From the web interface unable to save Hardware settings. I am able to restore settings and the settings will stick. workaround: modify config file with new settings then restore file

Pako2, 2018-12-30:
See [#192](https://github.com/esprfid/esp-rfid/issues/192).

msanchezt, 2019-01-03:
Had the same issue. As mentioned by @Pako2 this is solved in his fix.
Now running 0.9.1 and all working fine.

Thanks a lot @Pako2!

ccfiel, 2019-01-04:
@Pako2 @msanchezt no working with me. wemos d1, tried 0.9.1 and 0.9.4. I got this error

```
POST http://192.168.4.1/commit 405 (Method Not Allowed)
send @ required.js:4
ajax @ required.js:4
(anonymous) @ esprfid.js:1
(anonymous) @ required.js:4
each @ required.js:2
each @ required.js:2
(anonymous) @ required.js:4
i @ required.js:2
fireWith @ required.js:2
y @ required.js:4
c @ required.js:4
XMLHttpRequest.send (async)
send @ required.js:4
ajax @ required.js:4
n.fn.load @ required.js:4
inProgress @ esprfid.js:1
commit @ esprfid.js:1
onclick @ (index):112
```

closed 2019-01-10 by omersiar