omersiar/esp-rfid #217: Implement some changes

Maik2016, 2019-01-06:
Hello,

I have changed the 0.8 version to  2 new functions like   close and ring because my code lock also submits the ring command via wiegand.  
Also i found out, that there is no max retry limit available that makes brute force relay easy.
I also implemented the max retry with a blocking of the communication.
The coding is not finished ( not nice looking and not changeable through the WEB Interface as of now)  but works for now :D   If this is interesting I can try to upload the changes ( have to figure out how to do this)
Also I try to put this in the source code for 1.0 beta

omersiar, 2019-01-10:
2 seconds cool down between succeeding scans is just enough for the brute force.

omersiar, 2019-04-05:
@Maik2016 Any update on this?

Maik2016, 2019-04-19:
I have no update due to some work issues and traveling a lot :(  No time to look at it again. But end of next week I'm back home and hopefully have some time, I have to check what has changed in the source code till now.

omersiar, 2019-04-26:
@Maik2016 I'm preparing milestones for the v1.5, if you can just publish your changes through a fork or direct copy paste, community can work on it, no need to wait for polishing it.

Maik2016, 2019-04-28:
I createt a fork I hope this works, was the first time I used GitHub

omersiar, 2019-04-30:
Thanks for your contribution to the esp-rfid, much appreciated.