omersiar/esp-rfid #220: [ Frequently Asked Features ] Please read this before submitting new request

omersiar, 2019-01-10:
Thanks for using esp-rfid software.  Please do not submit your feature requests as issues and comment on this issue if you think your request is not in the list below instead.

- [ ] Wiegand Keypad Support
- [ ] Any Leds (Status leds, alarm leds, any visual indicators)
- [ ] Buzzers
- [ ] MQTT Subscribe or receiving commands

nardev, 2019-01-29:
Hi all,

There is important changes that i made, please review the explanation and tell me what do you think:

https://github.com/esprfid/esp-rfid/issues/132#issuecomment-458514199

Besides that, i have HTTP/MQTT full API also work in progress. Which enables full configuration and control of doors, buzzer and led and door status info. Perhaps that should be separated in another commit.

Also, i think that next upgrade should be possibility to extract working times from log's and possibility to interconnect devices and other controllers. For example to be able to open remote doors that are controller by simple cheap relay board. I think that would be amazing feature.

bib2, 2019-01-29:
I want to update my esp-rfid board from nardev. Whioch update file do i need for this from inside the zip-file?

generic.bin is for self-made rfid-boards?

Is forV2Board.bin the right file for me?

nardev, 2019-01-29:
@bib2 can you open new thread?

omersiar, 2019-01-29:
@bib2 forV2Board.bin is right.

omersiar, 2019-02-19:
#242

ITSanchez, 2019-02-27:
Hi, excelent project i like can configure a password for wifi on AP mode in GUI if principal wifi fail. 
Thanks

Batman313v, 2019-03-06:
I would love to see HID Support for this. The current sketch I wrote doesnt have a GUI or any way to configure options other than Changing the code directly

bib2, 2019-03-06:
It would be great, to get wifi RSSI information over MQTT.

So i could see, if the wifi coverage is good on the position i installed my esp-rfid. I have some sonoff-devices and there i get many information about the connected wifi.

Netoperz, 2019-03-29:
Hello, I have similar project in production, based on esp32, we do have whole mqttcommunication remote acess remote loggig and pincode propagation as well as the keypad support, time acess for the password/keypad, build in database, some ota functionality , etc. contact me if You are interested in cooperation, possibly we could share some code to improve your functionality ?

One question why not the ESP32 ?

nardev, 2019-03-30:
@Netoperz some libs don't support arduino esp32

Netoperz, 2019-03-30:
I understand. Depends on what is the problem, there are some ways to adopt almost any lib into ESP32. Whole idea is about how the Arduino/IO code is handled on the lower level. FOr example delay function does not actualy make delay, it calls vtaskdelay on the IDF Layer, so the RTOS can run other stuff, for example code that does not have delays is not "compatible" and when you add some the watchdog's a re fed right and the core functions like idle tasks can work their stuff and so on.

So if You know particular lib that is causing the problem let me know I may look at it.
And if You want to take a look at our code, i cannot push it directly to the open repo, so we need to speak on prv. 
I came here to use your code as verification software for my problems with wiegand keypads, i get some extra bits on D0 line, i i'm not sure why, is it hardware or the code, esp8266 is 5v tolerant so using this as wireless keypad adapter eliminates the need of logic level converters for d0 and d1 that i suspect are causing my extra bits.

Have you ever noticed some extra bits on viegad in Your tests ?

Netoperz, 2019-03-30:
@nardev @omersiar 
I'm thinking about adding the localization change option to the web UI, 
Before i will start doing something, how would You see the way it should/could be implemented ?
(The option for the user to choose the language of the UI to preffered for the user, english german,polish,spanish etc....
It would be needed for the first time use wizzard, that i also plan to add from my own project. 
I'm asking because if I will spare some time on coding it would be nice that what I will do would fit to the overall roadmap of the project.

omersiar, 2019-04-05:
@Netoperz localization should be done like this way:

* For WebUI we completely refactor HTML and Javascript for multi-language
* For firmware we can use something like https://github.com/jjssoftware/applicationStrings (log strings) or go with eventID approach.
* WebUI comes with English language by default (not on SPIFFS but on firmware level)
* Anyone can be able to upload their modified language file (to SPIFFS) then WebUI is get served with translated content
* So we do not care about translations are being part of firmware and can be installed, modified and corrected without reflashing the firmware.
* We can use one of the many translation services to produce quality translations.

arretx, 2019-06-25:
Okay, so the newest version is awesome, but now it opens up a whole world of things we need to be able to do.  For example, since I can send the 'opendoor' command via MQTT, I can de-activate the door lock, but only for the set time in the Web UI.  I need to be able to deliver the milliseconds for the lock in MQTT. 

To expand on that, if we were able to assign the amount of time the lock is de-activated based upon the RFID tag that's used to unlock it, that would be amazing.

One other thing.  The heartbeat that's sent tells HA that the lock is online.  A timeout can help HA determine if it's offline.  But, how would someone extract the simple binary value of "on" and "off" while the lock's relay is "on" or "off" for the state of the lock, versus the state of the door?

windy54, 2019-07-10:
Could we download the access log?
I have just taken over an implementation of this project in our Hackspace, it has been working fine for a couple of months.
I would like to be able to download the access log so we can see how often members enter the Hackspace. 
I am new to MQTT as well, is it possible to send a message every time someone enters the Hackspace? That way I could capture the metrics on another machine.

nardev, 2019-07-10:
Exactly, that is possible.

It's easier to get data through mqtt and real-time.

I would use free service like cloudmqtt for start. You would just need
another app on the other side that is subscribed to same mqtt topic and
that can save exact data, when someone gets in.

It's very very easy to make such service.

omersiar, 2019-08-26:
- [ ] Refactor Wi-Fi connection logic

omersiar, 2019-09-10:
#306

joelgamal, 2019-09-10:
Limited (scheduled) access for specific user

vrelk, 2019-11-05:
While I know some of these were listed as issues or comments above, I figured I would just add them to the list anyways since they are features of enterprise solutions. While I can't see all of them being needed for this, I figured I would list them for both you and others to see and comment/vote on.

- support for multiple fob types (HID corporate 1000, HID 34 bit, etc), or even better, custom options. See: http://cardinfo.barkweb.com.au/ (Only card number, and maybe facility code is important) **Personally, I would rank this as number one as I would like to use an HID ProxKey II or III and end up with a number matching what is printed on the fob.**
- Time based user roles (schedule for when user can open the door)
- Multiple doors (maybe 2 doors/readers per esp for this type of project)
- Hold door open during certain hours
- REX (request to exit) sensor (motion, or push a button).
- Using the REX, report if the door was forced open or not. Ex, someone used a key instead to unlock it, or forced the lock, etc.
- An alarm relay that triggers if the door is opened without REX. This could be motion, push button, or another reader on the other side of the door. OK, I know this one is a but outside the needs for this project, but the feature exists nonetheless, so maybe just MQTT if you implement a REX.

Drewi1904, 2019-12-10:
Hi at all
it would be great if you could control the unused GPIO via MQTT (for status LEDs for example)

devUBi, 2021-01-07:
Very nice project!
Would it be possible to add some http support to trigger an URL instead or in addition on switching the relay on and off?
Because then it would be very easy to control a shelly device directly with these URL's for example:
http://[shelly-ip]/relay/0?turn=on
and
http://[shelly-ip]/relay/0?turn=off

nardev, 2021-01-14:
@devUBi can this help you? https://youtu.be/m8uzLfFQQUk

devUBi, 2021-01-15:
@nardev thank you for pointing me to that video. Very interesting and of cause a solution but with the possibility of sending http calls in addition, you would be able to control the devices without any additional environment (mqtt broker f.e.)