omersiar/esp-rfid #223: after a while the RFID reader stops

donprosko, 2019-01-14:
1.0.0-rc1 (or better 0.9.4) on Lolin NodeMCU V3 with RFID RC522 and mosquitto version 1.5.5.
now MQTT messeages seem to be fine but after a while (sorry, don't know how long) the RFID reader stops. There is no reaction, no log entry and no line in there serial debug protocol. After a hard reset it works again until the next break.

araziforum, 2019-01-14:
I have same problem. I will try my chance with 0.9.5

omersiar, 2019-01-14:
There is no 0.9.5, and the version number is just a text, does not mean that the firmware does not include fixes.

Can you please give more details?

Is esp not responsive?

Can you login webui while rfid is not working?

donprosko, 2019-01-14:
The esp is full reachable with HTTP and everything seems to work. Only the reader is dead.

omersiar, 2019-01-14:
Then it may be a library issue. I will check the mfrc522 repo.

marceltech, 2019-01-15:
I'm facing the same problem, it seems that only sensor stop to work! I'm using RFID -RC522. Does it work with wiegand (https://pt.aliexpress.com/item/Boa-Qualidade-ZK-KR202E-Leitor-de-Cart-o-de-Controle-De-Acesso-RFID-Wiegand26-Leitor-de/32874980628.html) using password instead of rfid sensor?

marceltech, 2019-01-15:
Dear Colleagues, Just to add more details, the heartbeat mqtt message still receving that's make sense what Omer told, looks some issues with library.

donprosko, 2019-01-15:
maybe it makes sense to use the RST Pin of the RC522 again? Or if the RC522 itself works fine we can use the IRQ pin and trigger the ESP?

araziforum, 2019-01-15:
I'm testing. 40mins passed, still working fine.

araziforum, 2019-01-15:
2 hours, 40 mins passed still working. I think problem resolved. I will try it until next day.

donprosko, 2019-01-15:
Thank you Ömer. I've alread flashed the ESP with v1.0-hotfix1. after about 15 minutes the I rcognized no reaction on any RFID chip. So I seems form me the same behaviour as with 1.0.0-rc1.

omersiar, 2019-01-15:
Which rfid readers do you have?

donprosko, 2019-01-15:
the RFID RC522 from AZ-Delivery (only SPI mode possible).
So I set in Hardware Settings Reader Type = MFRC522.

marceltech, 2019-01-16:
> Can you guys test this https://github.com/esprfid/esp-rfid/releases/tag/1.0.1

Looks good now! Thanks!

araziforum, 2019-01-16:
16 hours, 9 mins passed works fine. Now I will cut the power and try to test again.

donprosko, 2019-01-16:
I asume I have troubles with the SPI bus. The RC522 should be outside of the house and the ESP inside. So nobody can manipulate the relay PIN from outside of the house. 
The SPI cable length have to be about 30 cm. That's a lot for the SPI clock speed of ...
#define MFRC522_SPICLOCK SPI_CLOCK_DIV4			// MFRC522 accept upto 10MHz
Is it possible to slow down the SPI clock speed to increase the SPI bus stability?

omersiar, 2019-01-16:
spi speed was hard coded to mfrc22 library, i guess changing it is just newly introduced.

you should try it yourself, cherry-pick the commit and set esp's speed as well.

78Ronin78, 2019-01-16:
How run project without device? I run web emulator, run localhost and get message: "Upgrade Required". How fix it?