omersiar/esp-rfid #226: Usage of Wire library.

Koepel, 2019-01-17:
In the file "esp-rfid/src/PN532.cpp" is a `Utils::DelayMilli(1);` (at the moment in line 950) when reading data with `I2cClass::Read();`. That delay may be removed.

The origin is at Adafruit, so I have made [this issue](https://github.com/adafruit/Adafruit_NFCShield_I2C/issues/23).

omersiar, 2019-01-18:
Thank you for reporting.

While it is based on Adafruit's initial code, Elmue did great job improving and polishing it. Please check it out.

http://netcult.ch/elmue/
https://www.codeproject.com/Articles/1096861/DIY-electronic-RFID-Door-Lock-with-Battery-Backup
https://www.codeproject.com/Members/Elmue

Koepel, 2019-01-18:
Thanks. I have written a message on his project page at codeproject. Maybe he will fix in a future version.
It is not a bug. It is only a small delay that is not needed.