omersiar/esp-rfid #238: config Settings

mekodak, 2019-01-31:
Hi

I am not able to change the Wi-Fi to Client,
When i try, and hit save.
I try to find it on my network, but nothing to find,
then i restart the power to the unit and check again, but still nothing to find.

And when i login to the unit as AP the settings is reset.

And i have tryed to use the "Test" button for the Relay test and nothing happens.

omersiar, 2019-02-01:
Hi,

Do you able to get information on status page? Which browser do you use?

mekodak, 2019-02-01:
It was no matter witch browser i used,
yes i was able to se the "Status site".

In Internet Explorer was i able to use the "test" button but when i pressed it it did not turn off again.

But the fail was that i was not able to scroll bovn om the "Modal Box" and i did not know that ther was a "Save & Reboot" button at the buttom.
But i did just Mark all the "Modal Box content" then it showed.

mekodak, 2019-02-01:
And by the way can i get a GPIO pinout of the board??

mekodak, 2019-02-01:
Some thing like:
[image]

nardev, 2019-02-06:
@mekodak  In fact, there is more in the docs here: https://github.com/esprfid/esp-rfid-relay-board

mekodak, 2019-02-06:
https://github.com/esprfid/esp-rfid-relay-board#relay
says that GPIO13 is used for relay, but how can it then be used for
MOSI (& for "D0") ("https://github.com/esprfid/esp-rfid#pin-layout")

And https://github.com/esprfid/esp-rfid-relay-board#reader-9-pin-terminal says that "D0" is GPIO4
I am a bit confused.

I want to test out with a "MFRC522".

omersiar, 2019-02-06:
You can not use mfrc522 with official hardware it is just for wiegand readers and wiegand bus is not Spi so there is no Mosi Miso

mekodak, 2019-02-06:
So the software is build for "MFRC522" without the board (only wit ESP8266)

omersiar, 2019-04-05:
Closing it for now, if you need additional help please open a new one.

closed 2019-04-05 by omersiar